prusa3d/Prusa-Firmware #1311: #define FANCHECK

swmarsh, 2018-11-06:
Having some issues with the #define FANCHECK.  I disable it as follows //#define FANCHECK which gives me the option to tell the printer if the hot end fan is spinning or the print fan is spinning like in the MK2s firmware. Only one problem.  It all work correctly just my heat bed starts to heat for no reason. I can not find anyway to disable the bed from heating when I disable this.  I have a 5v pwm noctua fan and a 2 wire blower for a print fan. Any clues??

3d-gussner, 2019-10-08:
@swmarsh  Long time no update here. Can you please close this issue?

Prusa-Support, 2022-06-02:
Hello! Sorry for intervening so late on this issue. Is the issue still actual in this specific case where you comment out the define FANCHECK? At this time, you can disable the fan check via printer menu: >menu >settings >fans check [on/off]. Of course disabling it at your risk and peril, overheating or simply heat creep can be an issue. 

Closing the issue for now given this info, but do not hesitate to let me know in case of more aspects to review, directly with me info@prusa3d.com mentioning me. 

Alessandro Pantaleo
Prusa Research

closed 2022-06-02 by Prusa-Support