prusa3d/Prusa-Firmware #1318: MK2/S firmware update

tonysmartantzoulatos, 2018-11-09:
Hello Developer Team,
I was told by support that this is the area to propose a firmware suggestion.
As i have been told you will release a new FW for the Mk2/S family and since also an MMU1 user i would really appreacate such an update. At the same time i wanted to ask if possible to have 2 versions of this FW. One for MK42 bed and one for MK52, since providing users the opportunity to upgrade to the MK52 removable plate would be really nice and not needing to do a "hotswap" of the bed after XYZ calibration and be forced to switch each time they need to calibrate their machines.

cimoalpacino, 2018-11-09:
Where have you heard about this update?

tonysmartantzoulatos, 2018-11-09:
@cimoalpacino where have i heard about a potential upcoming update for the MK2/S? I was talking on the facebook group with one of the people of tech support team and he told me that an update is coming. Since i told him about my request, he pointed me here to post it for the developers. I am sure both people and prusa would benefit from such a dual FW, since if they make it, they will be selling beds like hot cakes!

inquisitiveKangaroo, 2018-11-10:
Hi,
Thank you for the tip. The revitalized mk2s firmware update is in toDo list with medium priority and I hope we will get to it soon.

cimoalpacino, 2019-07-21:
i think this one can be closed

olafure, 2019-10-25:
Are there any news of this? 

The common firmware release between MK2s and MK3s was 3.2.1. 
Then came 3.3.0, which dropped support for MK2s.
(I have to add that my MK2s was 13 months old when Prusa dropped the firmware maintenance, and it has caused me quite a bit of frustration). 

Looking at the diffs between 3.2.1 and 3.3.0, I can't pinpoint a technical reason for a drastic measure such as end of life 13 months after I bought it.  

There are a lot of commits between 3.2.1 and 3.3.0.  The bulk of the size are the language additions. 

Release notes for 3.3.0: https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.3.0
Diff of 3.2.1 to 3.3.0: https://github.com/prusa3d/Prusa-Firmware/compare/v3.2.1...v3.3.0

This has particularly been made worse by suggestions that MK2S would be merged back on to the main release branch, like https://github.com/prusa3d/Prusa-Firmware/issues/1318#issuecomment-437624754

closed 2023-09-14 by github-actions[bot]