prusa3d/Prusa-Firmware #1325: Case Sensitivity for Commands and Filename

AustinSaintAubin, 2018-11-10:
I am having an issue with the Prusa Firmware.
When sending the print file command the I have noticed that the firmware is case sensitive.

# FAIL 
M23 PRINT~1.GCO
M24

# WORKS
M23 test~1.gco
M24

More details here:
https://github.com/luc-github/ESP3D/issues/273

rbm78bln, 2021-08-19:
Hey guys!

Any news on this issue?

I was trying with the current firmware 3.10.0-4481 on my MK3S and was also stumbling on this issue: M20 returns uppercase filesnames and M23, M30, and M32 work only with lowercase filenames.

To me it actually doesn't matter what IS the case, yet all commands should make use of the same case...

gudnimg, 2021-08-19:
It should be easy to convert the inputs for M23, M30, and M32 to lower case, thus any case would be supported but then there is no way to differeniate between two files with the same name but with different cases.

Regarding M20, I would expect to see the actual case on the file name in the same way as it is stored into memory. Whether it contains a capital letter, all upper case or all lower case. But if we do this I suppose we should maybe also make M23, M30 and M32 case sensitive?

But converting anything the user inputs into lower case will atleast improve the situation? Anyone have thoughts on this?

closed 2023-09-14 by github-actions[bot]