prusa3d/Prusa-Firmware #1327: MMU2, prusa MK 2.5 filament does not retract at multicolor prints

langph, 2018-11-10:
When doing a single color print, the filament ejects well after stopping the print. When I try a multi color print, it does not eject at the purge block, i have to abort the print.

inquisitiveKangaroo, 2018-11-15:
Hi,
Shall I understand it as the tip of your filament is unable to unload from the extruder during the color change?

langph, 2018-11-16:
Yes, it get stuck in the ptfe tube. Tried changing some settings, but no succes yet. Never had these problems with MMU1

stormzak, 2018-11-27:
Have the same issue, have you checked your Ramming settings.  Perhaps increase to 4s.

langph, 2018-11-27:
I tried that and eventually it did work, but the standard filament settings from slicr should work fine. I got the advice from prusa to put a little thermal paste between the cold end and the heatbreak (on the screw and then screw it in). I recently made a MK3 with MK2 parts, some new Prusa MK3 parts and a Bear upgrade and am using the MMU2 with it now. I used thermal paste, but do not know if it works because to many parameters have changed. I works flawless, however and the MMU2 motors work much better on 24 volts.

closed 2023-09-14 by github-actions[bot]