prusa3d/Prusa-Firmware #1336: Changing plates

Frigo007, 2018-11-13:
Hi,

When changing a sheet from a smooth to a Textured PEI, the height Z must change as each sheet is otherwise high. To change the height, you need to start printing and then change the height. For me, this is specifically for the smooth sheet -0.286 and for the Textured PEI plate -0.592. Therefore, it would make it easier to work if it was possible to select in the menu or shortcut what kind of plate was used.

Thank you

stahlfabrik, 2018-11-13:
Well. In the Menu you CAN configure live z even when not printing. That should be what you are looking for?

lokiidesigns, 2018-11-13:
I don't disagree with the sentiment of using Live Z height adjustment however if I were continuously swapping between smooth & textured plates I would want a stored value against each so that I can select the build plate when I swap them instead...

In the meantime it is a simple as recording the two Z Height values and changing this each time a swap is made but seems like a non-intuitive means to achieve this.

mcstoufer, 2018-11-21:
This appears to beg the questions of would we then need to track the z height on each side of a given plate independently? Or is the height delta of each side pretty close?

Mjankor, 2018-11-24:
This would be great. I running printbite on one side of a sheet, and pei in the other. They are quite different in thickness.

michalxfanta, 2018-12-06:
That's a good idea, thank you. We will consider its implementation in future releases.

GreenShadowMaker, 2019-07-27:
I look forward to having different saved live z heights for different sheets.

michalxfanta, 2019-10-11:
In firmware 3.8.0 we implemented Sheet profiles feature.

closed 2019-10-11 by michalxfanta