prusa3d/Prusa-Firmware #1337: Z Height Calibration Wizard

lokiidesigns, 2018-11-13:
Please look at flipping the direction of travel for the Z Height calibration so that it travels away from 0,0 rather than towards it. 

Main reason for doing so is that the user can actually see the filament being laid down without the hot end being in the way. Official guides explain that you are required to look at the profile of the extruded filament and this is difficult to do as it is without the hot end constantly in the way. 

It would also be useful if you could choose a slow, medium or fast travel speed as whilst there is ample travel to determine appropriate extrusion height it can go by pretty quickly - again though given it is difficult to see the profile due to the hot end this may not be needed.

inquisitiveKangaroo, 2018-11-14:
Hi,
thank you for a cool tip. The true is the First layer cal. should be as user friendly as possible and this sounds really promising.

ManuGithubSteam, 2018-11-27:
Good idea!

closed 2023-09-14 by github-actions[bot]