prusa3d/Prusa-Firmware #1341: why the x y and z axis is set to 16 usteps and the extruder is 32 usteps in FW.?

Crunch69, 2018-11-14:
why the x y and z axis is set to 16 usteps and the extruder is 32 usteps in FW.?  Wouldn't be better if all where set to 32 usteps

Crunch69, 2018-11-16:
So you are saying the Einsy board was antiquated the day they where sold no cushion was built in,  That makes no sense considering one of the main attributes of the TMC2130 stepper driver was the ability to have a high stepper resolution as much as 256 usteps

hoppke, 2018-11-16:
The trinamic drivers are nice, but the einsy is powered by an 8bit microcontroller, and you can't work around a limitation like that forever. 

Think about it this way: 
- all the supported LCD languages wouldn't fit at the same time until Prusa came up with a custom flashing method (and since that "upgrade" my MK3 boots slower).
- commit history shows how developers are spending time shaving a few hundred bytes of code/memory here and there (wouldn't be needed if not for the MCU limits...)
- the printer can't be fed more than ~14KB/s via USB - if you try to e.g. upload a .gcode file to SD via octoprint it could take almost as long as simply printing the part
- the part fan will whine when running at something other than 0% or 100% - again, linked to the processing power of the board
- the latest MMU upgrade comes with a board of its own, right?

It's not to say the einsy is bad, it's just that it's a dead-end choice if you insist on fitting the whole feature set of something like the MK3 onto it.