prusa3d/Prusa-Firmware #1342: long_pause should be accessible through G-Code

stuartpb, 2018-11-15:
[This function](https://github.com/prusa3d/Prusa-Firmware/blob/51b4858a8785d61c0e91ae65b29ee747efc258a8/Firmware/Marlin_main.cpp#L8186-L8210) is currently only triggerable by the LCD. It would be nice if we had a command (or an extended parameter to M25) to take these cleanup actions over USB (ie. from OctoPrint).

PavelSindler, 2018-11-15:
It actually is accessible. We have M601 for pause print and M602 for resuming print. It was originaly just for SD card printing. New pause print should work Correctly with Octoprint / USB printing. We will test different pause print scenarios with Octoprint tomorrow and let you know then.

PavelSindler, 2018-11-22:
Unfortunatelly our M601 / M602 gcodes works correctly just for SD card printing now. We will check how to  improve it. Currently the best option is to define gcode scripts in Octoprint.

closed 2023-09-14 by github-actions[bot]