prusa3d/Prusa-Firmware #1343: Lots of magic numbers in `pat9125.c`

AnHardt, 2018-11-15:
There are lots of magic numbers in `pat9125.c`.
I'd love to understand what's going on.
The official [datasheet](http://www.pixart.com/upload/PAT9125EL_GDS_V1.2_31052017_20171011150843.pdf) is not very helpful about the registers and their content.

>     // Now continues the magic sequence from the PAT912EL Application Note: Firmware Guides for Tracking Optimization.

Maybe that paper can enlighten me. Sadly i can't find it.

Can anyone publish a link or the paper here?

trevjonez, 2018-11-19:
there is some further info if you dig through this: https://os.mbed.com/teams/PixArt/code/9125_referenceCode/

closed 2018-11-20 by inquisitiveKangaroo

AnHardt, 2018-11-21:
Come on.
If `PAT9125_NEW_INIT` isn't poking purely random values to random registers someone had access to deeper knowledge. Where does it come from?

trevjonez, 2018-11-21:
I too would like to know where that info came from. I was just trying to add notes of further information I had found. It certainly wasn't everything 👍

hazcoper, 2023-11-01:
A couple of years later, but I would be very interested in that info as well