prusa3d/Prusa-Firmware #1345: 3.5.0-RC1 - MK3+MMU2 - Single filament Temp Wrong

TheNore, 2018-11-16:
After slicing a STL file and sending to octoprint via slic3r interface in MMU single mode:

1. The Tx command successfully executes, and filament is chosen correctly. 

2. The Tc command executes, the hot end heats correctly. 

     - However, the temp the hot end/bed are set to is 215/60. 
         NOT the temp specified int he Gcode when sliced - (PET/ABS + Others)
     - A Default temp is set at PLA. And the first layer is also at this temp. 
        Correct temp is then adjusted after the first layer is done. 

Temporary Resolution is to manually "tune" the temp for the appropriate filament range.

PavelSindler, 2018-11-19:
Start gcode sequence which was part of 3.5.0-RC1 release notes was just example for showing how to use Tx and Tc codes. It was example for PLA.

TheNore, 2018-11-19:
Thank you for clarifying, I see the S codes and will manually change them until SLic3r is updated.