prusa3d/Prusa-Firmware #1347: Remove bedtemp from activating safety timer.

Mjankor, 2018-11-17:
Not sure how feasible this is, but would it be possible to remove using the heatbed from the safety time.

I just replaced one of my PEI sheets with a new bed material that has to "set" for a few cycles at 110º for 1 hour. This required a bit of babysitting as the printer kept turning off the bed due to the safety timer.

However, that's a one-off, and I wouldn't request this feature if that was the only purpose behind being able to run the heatbed by itself. Having a temperature controlled plate can be useful for lots of purposes. They can be used for drying things out (eg: soggy phones), annealing of prints, even science experiments (like a hotplate in a lab).

PavelSindler, 2018-11-19:
You can use gcode M86 for changing safety timer settings. 
M86 - set safety timer expiration time in seconds; M86 S0 will disable safety timer

Babaorum163, 2020-06-21:
> 
> 
> You can use gcode M86 for changing safety timer settings.
> M86 - set safety timer expiration time in seconds; M86 S0 will disable safety timer

But how do you implement this code into the printer?

amatulic, 2023-06-05:
@Babaorum163 - you would send the g-code via Pronterface or Octoprint over a USB cable connected to the printer.