prusa3d/Prusa-Firmware #1349: Since FW 3.3.0 MK3 boots much slower and cpu performance acts like it is maxed out since LCD languages

Crunch69, 2018-11-19:
Since FW 3.3.0 MK3 boots much slower and cpu performance acts like it is maxed out since LCD languages.  Please fix the coding to restore the MK3 performance

Pathogenius, 2018-11-19:
I would love to have an English only version at the very least, even if it means I compile it myself, which would require instructions because I'm not very smart.

Crunch69, 2018-11-19:
An English MK3 version only.  Cutting MK2.5 code and languages off will allow a lot of needed room for improving the printer

Pathogenius, 2018-11-20:
I see there are 342 forks, so I wonder if maybe one of them involves exactly what you speak of?

Pathogenius, 2018-11-20:
Well looky here! These instructions are even simple enough for me! I'm hoping to explore these options that you mentioned Crunch69, and thanks for the inspiration! GOGOGO!

https://github.com/sorcererlc/prusa-firmware-bear-mk2.5-3

closed 2023-09-06 by github-actions[bot]