prusa3d/Prusa-Firmware #1358: Add other material to first layer calibration

duartemv, 2018-11-23:
Hello,

I have an original Prusa i3 MK3.

During the first layer calibration test, the user is asked if there is PLA in the printer. I personally don't use PLA, and am forced to manually change the nozzle and bed temperature for ABS during the heat up phase.

The user should be asked what filament he/she is using and then the printer would heat up for that material.

This is basically and enhancement request.

Thank you,
Duarte Vasconcelos

inquisitiveKangaroo, 2018-11-25:
Hi,
We are planning to add this feature into the menu. For now I can share you gcodes to make it manualy.

Just change the ending **.txt** to **.gcode** to make it work.) 
(GitHub won't let you upload .gcode...)

[V2Calibration_ABS.txt](https://github.com/prusa3d/Prusa-Firmware/files/2612760/V2Calibration_ABS.txt)
[V2Calibration_PET.txt](https://github.com/prusa3d/Prusa-Firmware/files/2612761/V2Calibration_PET.txt)

closed 2018-11-25 by inquisitiveKangaroo

mszeemann, 2019-03-05:
If you plan to add different materials please also think about adding different nozzle sizes. For example we work with a 0,6mm nozzle. So doing the build-in first layer calibration is useless. Having the ability to choose between PLA / PETG / ABS as material **AND** 0,25 / 0,40 / 0,60 as nozzle should fit 99,9% of the use cases.

servel333, 2019-05-21:
What is needed is for all related settings to be available when running the first layer calibration, or available as separate settings in the settings page. This makes the UI a lot more complex though, so maybe we need custom materiel profiles that can be selected when running the first layer calibration.

michalxfanta, 2019-10-11:
In firmware 3.8.1 RC1 we implemented this feature.

diogorsergio, 2019-12-22:
> In firmware 3.8.1 RC1 we implemented this feature.

@michalxfanta MK2S Firmware is 3.2.3 any change of this making into that?