prusa3d/Prusa-Firmware #1359: X/Y crash recovery too sensitive

xrm, 2018-11-23:
Hi there,

Prusa i3 MK3 with MMU2 and FW 3.5.0.

When my printer detects a skip in X and/or Y and tries to re-home, it fails to find the real home position and stops midway (for example at x = 3 cm), resulting in a different origin (and thereby wrecking my print).

The issue only shows up during the crash detection; going to 0/0 at the beginning of a print works flawlessly. Belt tensions are normal.

Could it be that the recovery doesn't use the same high power settings as the initial calibration does?

Thanks
Sebastian

maximlevitsky, 2018-12-09:
Please check that noting blocks the X axis from going to a hard stop at origin.
This means that when you manually move the extruder to left, it eventually stops and you see that its touches the left X motor holder plastic part and not for example the zip ties on the back touch the einsy case or so.

The zip ties on the extruder and the hot end wires are the main culprit for that.

xrm, 2018-12-10:
No problem there. It works with G28 or any other "go to 0/0" commands, it only fails during the crash detection.

I also know how to circumvent this, but I would expect the printer to behave the same way in both situations.

haarp, 2018-12-27:
Are you sure the extruder cable isn't pushing the X carriage away from X=0? As long as the stepper is active, it can hold the carriage there, but when they switch off, the elastic compression of the cable can push it a bit to the right.

GurliGebis, 2018-12-27:
@haarp If the cable is getting in the way, it sounds like it is assembled wrong (I did so myself, I forgot to put the extruder motor cables in behind the extruder body where they should be) :)

xrm, 2019-01-03:
It happened during the movement and if it reached 0/0, it also stayed there even if/when the steppers were turned off.

jrmiller82, 2019-02-28:
Similar problem here; except the auto-home only works on initial calibration for the x-axis i.e. running the system wizard. I'm on a new kit mk3s. The x carriage moves freely, the belts are properly tensioned, but if I choose auto-home the x carriage only moves maybe a 1cm to the left no matter where it starts. So, if the x carriage is at x=3cm, it never re-homes back to 0 and the printer thinks that x=3cm is the new home. 

The only solution I've found is to literally power off the printer, manually move the x carriage all the way to the left (i.e. bump into the x-stepper-motor housing) and then choose auto home. 

There are no obstructions to the x-carriage and it doesn't matter where on the x-axis the x carriage is, the same behavior occurs.

Prusa-Support, 2022-06-24:
This issue could be closed because stale, but I'm wondering if the problem persists and if you are able to reproduce it.

It was mentioned that the X-axis may home 3 cm away from the beginning of the X-axis. How consistent is such offset between the homing attempts?
That's surely interesting how standard auto-homing doesn't fail in the same way.
Videos and pictures can help understand and evaluate the problem.

Off the top of my head, I could think that the distance the X-carriage travels affects the way it tackles resistance towards the beginning of the X-axis; or that one linear bearing is partially damaged, causing a sporadical stall of inner spheres and a crash in the first place, while the incorrect homing is yet another consequence of the inner spheres that lose alignment/rotation.

Maybe, a replacement of bearings and rods would do the trick or, as mentioned in issue #1096, this issue is due to external factors or HW/assembly related inconsistencies (check the relevant [comment](https://github.com/prusa3d/Prusa-Firmware/issues/1096#issuecomment-1165799584)).

So far, I'm not sure how reproducible the issue is the concern on the firmware side is not confirmed.
Hopefully, problems like these can be resolved with a proper troubleshooting session, so please consider reaching out the Technical Support before posting new similar problems.
https://help.prusa3d.com/article/customer-support_2287

Michele Moramarco
Prusa Research

xrm, 2022-07-06:
I can currently no longer reproduce this. I am not sure if this improved after cleaning the rods and bearings or if this resolved with a software update (I didn't print for a few weeks and did a complete overhaul before I started printing again).

As I wrote, I was able to circumvent this, the only reason for this ticket was that the way the printer moved during a crash-recovery seemed to be different from the "usual" one which was unexpected.

If I encounter the issue again, I'll get in contact.

Thanks!

(Ticket can be closed from my side if there are no other open question and if no one else is able to reproduce this (@jrmiller82 seems to have had the same issue?).)

closed 2022-07-25 by xrm