prusa3d/Prusa-Firmware #1363: M600 filament change should prompt user for MMU extruder to use.

mgg4, 2018-11-25:
Printer: Prusa i3 Mk3 with MMU2.0

When using the printer in MMU Single Mode, and configuring a color change using ColorPrint, the M600 command is properly inserted.  When the resulting G-code is processed, the startup up includes prompting the user to select which extruder to use.  The first color prints normally.

When the M600 command is processed, the print head parks, and the filament is unloaded by the MMU.  Resuming the print will reload the same extruder to the nozzle, with no opportunity for the user to select a different extruder to use for the next color.  Even if the MMU is manually set to use a different extruder, when the print is continued, the MMU moves back to the extruder selected at the beginning of the print job.

This means, in order to actually change colors, the filament must be completely removed from the MMU, and the new filament inserted into that extruder path on the MMU.  This is cumbersome and completely unnecessary.

At line 572-573 in mmu.cpp, there is code to ask the user if they want to switch to a different extruder.  If the answer is yes, then the choose_extruder_menu() function is called again to select a new extruder.  However, these lines are inside an #ifdef block, with a tag of MMU_M600_SWITCH_EXTRUDER.  This tag is not defined anywhere else in the code base, effectively commenting out lines necessary to switch the print to a different MMU filament path.

There may be a reason for disabling this feature.  If so, please correct the other issues so this feature can be re-enabled.  If this was merely an oversight, please either remove the #ifdef block around this code, or properly define the MMU_M600_SWITCH_EXTRUDER flag in mmu.h.  I would be happy to provide a pull request for this change, if you'd prefer.

darBis, 2019-02-02:
Just run into same issue today. :-( If the printer would keep MMU head in place, that way one could switch filament manually before loading and resuming the print.

Menu on display would be nice, but at least please allow to continue with different "current" MMU position.

TheNore, 2019-04-09:
Also experienced this today as well.

lukedubber, 2019-04-23:
Mine is the MK3S & MMU2S and have the same issue. Really a bummer, I love multi color prints that the MMU2 offers but I sometimes just want to do it the old way with M600.

gwaland, 2019-05-27:
2.5s/MMU2s upgraded from an mmu1 here and just hit this issue as well.  Very disappointed to find this is the case after trying the feature using prusaslicer.

xiphmont, 2019-05-29:
Uploaded pull request #1881 to implement this feature.  Using it here, works as expected (though it does not fix any other minor MMU extruder switching bugs ;-)

gwaland, 2019-05-29:
Thank you! I've grabbed your branch and I'll try to test it out tomorrow.

darBis, 2019-06-06:
I tried the patch applied to the official 3.7.1 branch on my MK3S + MMU2S today and it worked like a charm.
Thanks @xiphmont :-)

hypercolor916, 2019-09-14:
I applied  the patch from @xiphmont to official 3.8.0 and it worked perfectly on my MK3S MMU2S. Hopefully PrusaSlicer's Color Print feature will be able to assign different extruders someday...
Thank you!

bujocek, 2020-01-09:
Actually there is (still very cumbersome and annoying) workaround to get the filament changed during the print. Here are the steps:

1. wait till the print pauses the print for filament change
2. confirm the filament change and let the printer load the SAME filament back into extruder
3. when the printer tries to continue quickly pause the print
4. using the menu unload filament from extruder
5. load another filament into extruder (the one which you want to continue the print with)
6. resume the print
7. Now the printer tries the filament change process again so just confirm it. It will unload and load back the current (newly selected) filament.

I hope this helps those of you who cant or dont want to change your firmware.

zostay, 2020-02-11:
Having the MMU and being unable to switch filaments while paused without resorting to pausing again right after it already paused the first time is really frustrating. I would love to see this fixed.

darBis, 2020-09-14:
If anyone is interested, I made patched prebuilt firmware from 3.9.1-RC1:
https://github.com/darBis/Prusa-Firmware/releases/tag/v3.9.1-RC1-M600p
I have tested it on my MK3S and works fine.

JamesPatageule, 2020-09-18:
Also very frustrating by this little thing ...
Hope will be released soon in OFW !

sioux612, 2020-10-13:
Has this been fixedor is the Patch by darBis the only way to make that work? Anybody else test that FW? I'd prefer not to move to a possibly unstable RC version when OFW is available as well.

gwaland, 2020-10-13:
As far as I'm aware they aren't going to do the patch.  They did update the prusa slicer so you can change filament by layer height instead of just getting the m600 command so I've been using that as my work around.

darBis, 2020-10-17:
> @sioux612  
> 
> Has this been fixedor is the Patch by darBis the only way to make that work? Anybody else test that FW? I'd prefer not to move to a possibly unstable RC version when OFW is available as well.

Hi, I am using it on my machine without issues, but I understand your worries.
I am going to publish patched 3.9.1 soon.

Just that there is no doubt, the patch was created by @xiphmont, I only applied it to a more recent firmware. I don't want to steal any credit. ;-)

darBis, 2020-11-02:
Patch applied to 3.9.1 and prebuilt for download here: https://github.com/darBis/Prusa-Firmware/releases/tag/v3.9.1-M600p
Tested on my MK3S.

salmander, 2022-02-27:
We are on v3.10.1 and this is still not merged with master?

leptun, 2022-02-28:
We will add this in the new MMU fw implementation. It will come in fw 3.12

DeanDavis, 2022-07-14:
Should just ask which extruder to use. No matter which extruder is chosen the filament from current extruder is unloaded, the selector moved (or not if chosen extruder = current extruder) and filament loaded.

rtyr, 2022-07-14:
Do not forget that M600 command could be also used with regular MMU print (it is often used to extend maximum possible colors on one model), you should not allow tool change via M600 command on such print (otherwise it will likely fail).

gudnimg, 2022-09-14:
Just for future reference, this feature request was implemented recently here: https://github.com/prusa3d/Prusa-Firmware/commit/652ede3480b1a3e51c19a9a80911a6fb20b382fe (`MK3_MMU2` branch)

gudnimg, 2023-07-31:
Closing as this is now resolved with the release on FW 3.13.0 and MMU FW 3.0.0 :)

gwaland, 2023-08-13:
Any ETA for those of us with a mk2.5s?