prusa3d/Prusa-Firmware #1364: Modified print speed is reset to 100% every filament change with MK3 MMU2

rncfiles, 2018-11-26:
Modified print speeds either from the info screen or in the "tune" menu does not persist through filament changes with MMU2.   At every filament change it resets to 100%.   This is with MK3 FW 3.5.0

I verified that at least one other "tune" option (nozzle temp) does not reset, so it could just be print speed that behaves poorly.

PavelSindler, 2018-11-26:
Yes, Slic3r sets speed to 100% before filament change (there is gcode M220 S100 for it). We have encountered similar issue on Multi Material version 1 and it is solved there in both firmware and Slic3r by gcodes for speed backup/restore. But unfortunatelly on MK3 + MMU2 both Slic3r and fw does not support these gcodes yet. I will fix it in firmware soon.

Hadeshoof, 2018-12-08:
This is a huge problem when printing PETG as 100% speed basically fails every single print using the MMU2.  Since PETG is the only thing I use, this MMU2 is worthless to me until there is a workaround or fix.

michalxfanta, 2018-12-15:
Duplicate #1208

AbeFM, 2018-12-21:
Certainly not every change. It will go to a speed, but not always 100%. I've seen a lot at 35%, or anything in the 30's.

ColoradoMTBer, 2019-03-04:
Has this print speed issue been fixed yet with the latest FW version 1.42.0-alpha7 or FW version 1.41.3? It doesn't seem like it has. This is really a problem printing with PETG. The on-the-fly print speed adjustment using the knob on the printer interface makes fine tuning speed so easy. I'm not convinced that the print speed settings are even adjustable using slic3r as they seem to do exactly the same thing as any the on-the-fly settings and revert to 100% after the first filament change...

TickTock-, 2019-04-07:
Also, extruder temp reverts back to original temperature next time that same filament is loaded during the print. 
I noticed an earlier comment says otherwise so, perhaps, the difference is if the two sources have the same extruder temp.  For me, I was printing PETG with PVA for supports.  After tweaking the PVA extruder temp, it switched to the PETG as normal but when it went back to PVA, the tune setting was lost.  Tune needs to store independent values for each of the five MMU sources.

mkbel, 2019-12-25:
@rncfiles This is issue of gcode generated from PrusaSlicer. It is supposed to be already fixed. Can you retest with latest PrusaSlicer so we can close this?

AbeFM, 2019-12-27:
Certainly watching this one - I've always just given up and resliced if I
have different temperatures....

3d-gussner, 2020-07-13:
@rncfiles Is this still an issue? If it has been fixed in a newer firmware version, please close this issue.
[stale]

JohnnyDeer, 2020-08-12:
I'm closing this issue as stale, thanks for understanding.

closed 2020-08-12 by JohnnyDeer