prusa3d/Prusa-Firmware #1368: Nozzle temperature goes down and printer pause waiting the temperature raise again

bionno, 2018-11-29:
Hello, 

sometimes the printer pause printing because the nozzle temperature goes under the settings i've done (for instance i set the PLA to 206 Celsius and sometimes goes to 195 .. pause .. heating again ... when reaches 206 resume with little blob of filament) .

i've used the other 12V +/- on the Power Supply , connected to a DC 5V converter to supply the Raspberry with Octopi and 4 USB led stick .

Yesterday i've done the PID calibration and today i'll retry some print , maybe the problem was the PID calibration ... maybe the power consuming of Raspberry (i hope not because is only 3A/ 5V )  ... or maybe the Formware ..

PavelSindler, 2018-11-29:
Which printer version and firmware version do you have?

bionno, 2018-11-30:
Anyone tried out  this issue ?

bionno, 2018-12-01:
TOday the printer give me the error " PREHEAT ERROR" ...  do you think it is a power supply problem because the DC converter and Raspberry PI3 with led light connected directly ?

Now i ve removed 2 led light from the Raspberry USB , hope this was the problem ..

bionno, 2018-12-01:
MAybe this problem is related with this post "

https://github.com/prusa3d/Prusa-Firmware/issues/1362

joseph-greiner, 2018-12-02:
I can confirm the original issue is happening for me. Firmware 3.5.0-1749, MKIII. No raspberry Pi, no accessories except the E3D silicone sock. Sometimes, not always, when the nozzle temp drops below the set temperature, when doing solid infill on my particular print, the printer will pause. Set temp is 250C, when the nozzle temp drops the print pauses and "heating" is displayed on the screen. I have never watched the temps before so I don't know if the temperature variation is new to this firmware. But the pausing definitely is. I am using the latest software package as well, Slic3r 1.41.2 win64.

Here are a couple videos of it pausing.
Pauses at the beginning and towards the end, shows the temperature screen.
https://www.youtube.com/watch?v=JKRrIBc07ME (2min)
Here is the main screen:
https://www.youtube.com/watch?v=SPiu-Dl-afs (20 sec)

ThomasRLarsen, 2019-01-30:
I have the same problem. Nozzle temperature drops, and the printer sometimes pauses to heat the nozzle. Using Firmware 3.5.1

[image]

bionno, 2019-02-02:
@inquisitiveKangaroo has closed the referred issue (https://github.com/prusa3d/Prusa-Firmware/issues/1362) so the new firmware upgrade , available in a week , will resolve the issue .

Thanks.

closed 2023-09-14 by github-actions[bot]