prusa3d/Prusa-Firmware #1370: Firmware 3.5.0 - issues after cancelled print

insanity67, 2018-11-30:
Hi,
if I pause a print, then cancel it and run the same print again without resetting the printer, it somehow skips the purge line. The printer moves above the spot, but no purge line is extruded. It begins to print the new skirt, and after some time filament is coming out of the nozzle.

I´m on the 3.5.0 from Prusa downloads with stock MK3.

Best regards,
Bastian

inquisitiveKangaroo, 2018-12-04:
Hi,
did you performed the bowden calibration for each filament? Simillar thing happend to me too. Sometimes a string remains in the bowden and the filament has higher friction then. I set up my bowden cal. slightly under the bontech and since then it has never happened again to me. Sometimes it purge more than neeeded into the wipe tower but prints became much more reliable.

insanity67, 2018-12-06:
Hi, no I did no calibration, I dont have a MMU(2).

mujido, 2019-02-09:
I have also seen this with the mk3 and 3.5.1 firmware. Seems like it does an excessive retraction and takes a while to get material back out.

closed 2023-09-14 by github-actions[bot]