prusa3d/Prusa-Firmware #1371: Lin. correction values are lost after reset

knoopx, 2018-11-30:
They are not properly saved to eeprom (broken since 3.4.1)

zoltanf, 2018-12-07:
Same here, printed the ECOR tower and lowest (off) was best. I presume they will remove this menu item and the whole concept of linearity correction from future firmware versions.

hoppke, 2018-12-07:
Might be valuable to people who run prusa machines with non-stock setups though. Probably no need for it in the lcd menu, but as a firmware feature it's still nice to have (when it works).

aemileski, 2019-01-13:
Found the problem.  The code checks menu_leaving but not LCD_CLICKED.

Bad code:
[code]ultralcd.cpp 4734: if(menu_leaving)[/code]

Good example:
[code]ultralcd.cpp 2875:  if (LCD_CLICKED || menu_leaving)[/code]

aemileski, 2019-04-12:
Still a bug in 3.7.0 released 2019-04-01

Seriously?

```
ultralcd.cpp 5170:	if(menu_leaving)
ultralcd.cpp 5171:	{
ultralcd.cpp 5172:	    lcd_settings_linearity_correction_menu_save();
ultralcd.cpp 5173:	}
```

GurliGebis, 2019-04-14:
@aemileski may I suggest you create a pull request for it?
Since it is the official way to submit changes, it might get them to merge it (having it where it should be, instead of them having to dig through issues for code changes) :)

3d-gussner, 2020-07-13:
@knoopx Is this still an issue? I think it has been fixed in a newer firmware version, please confirm and close this issue.
[stale][fixed]

knoopx, 2020-07-13:
@3d-gussner I don't know, I moved to klipper years ago because of the instability and lack of open development and never looked back again.

3d-gussner, 2020-07-13:
Good luck with Klipper sounds very interesting.🤞🍀
Can you please close this issue? Always nicer if the owner close issues instead of a forced close.😉

aemileski, 2020-07-13:
@knoopx @3d-gussner This was finally fixed in v3.7.1 (May 13th, 2019) 

Note that open issue #1585 is a duplicate (I also reported that Mar 1st, 2019).

Yes, please close them both.

closed 2020-08-12 by JohnnyDeer