prusa3d/Prusa-Firmware #1375: MMU2 / Slic3r / Mk2.5 - Wrong filament printed for several layers, no indication of this in Slic3r preview.

domesticatedviking, 2018-11-30:
Slic3r 1.41.1+linux64,  Ubuntu 18.04.1LTS 
Printer:  MK2.5, Firmware 3.5.0-rc1-1688, MMU2 firmware: 1.0.2-132

Filament #3 was selected for several several layers instead of filament #4.  No indication this would happen in Slic3r preview.

Zipped 3mf
[shield setup.zip](https://github.com/prusa3d/Prusa-Firmware/files/2635210/shield.setup.zip)

**Print setup (red lambda objects configured as support blockers)**
[image]

**Model as sliced:**
[image]

**Printed output:**
[image]

inquisitiveKangaroo, 2018-12-04:
Hi, thank you for your report.
It seems like the red wasn't purged properly. Is it possible to retry to slice it in 1.41.2 and print it on fw 3.5.0 - 1749 and share the result. Or to share the stl? .)

domesticatedviking, 2018-12-04:
This is not a purge issue.  The color was wrong for several layers, each of which had material changes.

closed 2023-09-14 by github-actions[bot]