prusa3d/Prusa-Firmware #1376: Leveling Interpolation Error

rwensley, 2018-12-02:
A close examination of many returns by G81 shows that the interpolated values range from slightly non-linear to grossly non-linear.  As these values are used to adjust Z during printing an issue may arise.
Attached is a screenshot from excel of a G81 response and a graph showing the non-linearity.

[image]

rwensley, 2018-12-05:
Looking at the algorithms available in the existing source, both of these calculations have errors in them.

A simple linear interpolation would work fine here.
Given X0 - X6 are known positions as well as Y0 - Y6
and
Z0, Z3, Z6 are known for each Y row

Then solving for Z(i) = ((x(i) - X0)*((Z3-Z0)/(X3-X0))+Z0 for X(i) < X3
or
Z(i) = ((x(i) - X3)*((Z6-Z3)/(X6-X3)+Z3 for X(i) > X3

Solving for X1, X2, X4, X5 for Y0, Y3, Y6 first then solve for the remaining values in the Y direction.

I am not a programmer so this is a far as I can go. 
I hope this helps.

rwensley, 2018-12-05:
Opps...Closed it by mistake.  This is a serious issues and needs attention. IMO

Pathogenius, 2018-12-05:
This is interesting to me because it can explain some otherwise inexplicable anomalies in my bed mesh leveling attempts, specifically large sections of poorly leveled areas appearing after entering mesh correction values. I had previously assumed this was caused by Pinda/magnet errorsm but this explanation seems to fit better.

rwensley, 2018-12-06:
So here is some additional data from older versions of firmware and other users. This confirms the issue.
[image]
[image]

rwensley, 2018-12-06:
I failed to take the thermal sag of the heated bed into account.  It seems that the interpolation should NOT be linear and is likely correct as is.
Closing this issue.

closed 2018-12-06 by rwensley