prusa3d/Prusa-Firmware #1377: [3.5.0] Idler body motor stops responding

Flav1u, 2018-12-03:
Slic3er PE Version: 1.41.2
FW MMU2: 1.01
FW MK3: 3.5.0

It happened several times during long prints (after 8+ hours) that suddenly the stepper motor for the idler-body is “dead”. It doesn’t react anymore, and I can rotate it easily by hand (= no current?). After restarting the printer, it works normal again.
This never happened with previous firmware versions, only with 3.5.0.

Panayiotis-git, 2018-12-03:
Same thing happened to me. I noticed it after hours of print, on a filament change, the MMU unit was not doing anything. I also could rotate it freely by hand.

PavelSindler, 2018-12-03:
We are working on this issue. From our tests it seems, that it is not firmware related, but it is caused by mechanical contact / cables which provides power supply for MMU2. We are wotking on detection that this happened and automatic rehoming of axis.

TheHammockGuy, 2018-12-04:
I have also experienced this issue.  As it wasn't explicitly stated by others I wanted to add that this doesn't only happen to the idler motor.  I have also seen this happen with the selector and extruder motors on the MMU.  It can also happen to any combination of the three motors.

TheHammockGuy, 2018-12-04:
> 
> 
> Same thing happened to me. I noticed it after hours of print, on a filament change, the MMU unit was not doing anything. I also could rotate it freely by hand.
> I managed to recover by resetting only the MMU unit.

Is there a button combination on the MMU that you used to do this reset?  Or did you physically unplug and plug the power to the unit to recover?

TheHammockGuy, 2018-12-04:
Have you identified which contacts are an issue? And if so is there anything that can be done to said connection as a mitigation until the firmware is updated to detect, and recover from, this issue?

PavelSindler, 2018-12-04:
We don't know which contacts cause the issue yet. It seems it is somewhere in power supply cables, maybe contact on Einsy board, but we have just one printer where we can easily trigger this issue, so we do not want to fix it as we need the issue for our fw development :-)

Panayiotis-git, 2018-12-04:
@TheHammockGuy  on the right side of the mmu unit, there is a small reset button next to the usb connector.

Panayiotis-git, 2018-12-09:
I have started a 17 hours print. Till now, after 8 hours and 10 hours remaining, it has already happened 8 times...

And when this happens, all motors are released and I can rotate them freely. Before I start, I have checked for loose connections or bad soldering and I haven't found something.

Is there any power saving feature that releases the motors?

Pathogenius, 2018-12-09:
Has anyone experiencing this issue pointed a temp gun around at wires possibly identifying a weak link, since weak link can boost resistance causing voltage drop causing current rise causing heat. Just a shot in the dark.

Panayiotis-git, 2018-12-10:
The printing finished after 20h30m and this problem appeared more than 30 times during the printing.

I have to say that this time I was in custom builds on both MMU unit and MK3:
MMU: git commit 2b98387
MK3: git commit  9087645

Previously, I have done a lots of prints using the official MMU2 rel. 1.0.2 and MK3 custom build based on 3.4.2 and this problem was very rare.

@Pathogenius I've checked for hot areas on the board, the motors and the cabling by touch and during the operation everything was running cool.

Flav1u, 2018-12-12:
I changed all the MMU PTFE tubes to ones with 2,5 mm id. So the whole system has now way less friction. I’ve printed a 14 h print and a 18 h print and no issues. Of course I don’t know if this is a coincidence.

NatDenkin, 2018-12-13:
I have the same problem with an MK2.5+MMU2 running  standard builds 3.5.0-1749 and 1.0.2-132.

Panayiotis-git, 2018-12-19:
After some prints I would say that this issue has been fixed with the combination of MK3 3.5.0 and MMU 1.0.3.

If I only haven't to throw away all of these prints (about 12 models) because of skipped layers...
I mean, even this issue with the paralyzed unit could be bypassed by resetting the MMU unit and follow a specific procedure to recover the print.
But the incapability of the system to detect that the filament has stuck to the orange tube and is not going to ever reach the extruder but the printer continues to print with air... 

It is called complete loss of functionality and is consider as a blocking issue.

msallena, 2018-12-21:
I've updated to 1.0.3 and 3.5.1 firmware, and still see this issue - it just started happening for me a couple of days ago - not really related to any firmware upgrade.

MrKrankor, 2018-12-25:
I am also having this problem recently.  1.0.3 and 3.5.1, but I also had it a few times with 1.0.2 and 3.5.1

It seems to be somewhat random.  Sometimes I can get through 100 filament changes, other times I am having to reset it every 4 or 5 changes.  

And it affects both the idler selector motor and pulley motor.  Sometimes one, sometimes both.

Panayiotis-git, 2018-12-28:
Last days, I tried to figure out what is the root cause of this problem. I found that if I pushed the power or serial comm connectors to a specific direction (while they were plugged to the MMU's PCB), then the motors stopped responding. It was a repeatable behavior.

I then checked if the PCB connectors were not soldered properly. I re-heated/re-soldered them, but the issue was still there.

So, I assembled my own cables and tried again. The new cables plugged firmly to the PCB connectors. After the cable replacement, I cannot anymore reproduce the issue.

I believe that the problem is the contact of the connectors. It is somehow loose. As the cables are moving or vibrating during the printing, it happens that a signal or the 5V/24V/GND connection is momentary loosing contact and somehow the motors are deactivated.

AbeFM, 2019-01-22:
I gave up talking about that exact behavior before this thread happened. 
I had the exact thought/conclusion - bad connectors caused momentary lapses. Pulling the cables tight kinda helps prevent motion, but you get glitching.

It's really nice to see someone else verify this, so thanks for posting!!!!

Panayiotis-git, 2019-01-23:
Here in Europe, I find easy to source latched connectors for the MMU side from www.tme.eu:
https://www.tme.eu/en/details/mx-70066-0178/raster-signal-connectors-254mm/molex/50-57-9404/
https://www.tme.eu/en/details/mx-70066-0179/raster-signal-connectors-254mm/molex/50-57-9405/
https://www.tme.eu/en/details/mx-16-02-0088/raster-signal-connectors-254mm/molex/16-02-0088/

For the MK3 side I had some spare 5-pin female connectors, like these:
https://www.tme.eu/en/details/nsr-05/pin-headers/ninigi/
I find them at the local electronics shop.

nullenvoid, 2019-04-07:
I can confirm that changing out the molex connections resolves the issue.

TheHammockGuy, 2019-12-17:
@PavelSindler Is there still any effort being made to address this issue?

tdub415, 2020-03-19:
I'm experiencing this issue with increasing frequency.  Is there anything more being done about this issue?

ALittleSlow, 2020-12-19:
I am experiencing this issue as well. Sometimes it shows up as motors stalling against the hard stops instead of detecting them. Tuning the detection current limits doesn't help. This happens mostly with the selector motor. Sometimes the motor just refuses to move when it's supposed to. Sometimes wiggling the wires fixes it. Usually resetting the MMU fixes it. Happens on all three motors, but usually the selector. Seems to be happening with increasing frequency. I did find a loose pin on the selector motor Molex connector which I was able to secure. But the problem has repeated since fixing that. I have removed and re-secured the power leads at the ring terminals on the Einsy board. I have monitored the voltage at that point and it's a pretty consistent 23.74+/- 0.06 V. Very occasionally going down to 23.5 V. I have checked the resistance of the four power conductors and they are all similar and all around 0.3 ohms per wire, which isn't much over the contact resistance of my ohmmeter leads. I have replaced the MMU2S PCB (accidentally damaged the first one) to no effect.

Prusa-Support, 2023-04-07:
Our Help section kept growing and improving over time, and it will continue so. 🙂 
I'll include a few possibly relevant and useful links.
https://help.prusa3d.com/article/mmu2s-leds-meaning_2187
https://help.prusa3d.com/en/article/mmu2s-idler-unable-to-move-freely_1996
https://help.prusa3d.com/en/article/mmu2s-selector-not-moving_1997
https://help.prusa3d.com/en/article/mmu-loading-failed_2135
https://help.prusa3d.com/en/article/mmu2s-setup-and-inspection_2233
https://help.prusa3d.com/article/factory-reset-mmu2s_1920

Also, the MMU firmware has been totally rewritten and a new user-friendly version is expected to be released sometime soon. It will not only be more robust but also greatly help the troubleshootings.

Is this issue still relevant?

Michele Moramarco
Prusa Research

closed 2023-08-01 by Prusa-Support