prusa3d/Prusa-Firmware #1378: [3.5.0] M601 results in an infinite pause/resume loop

TheHammockGuy, 2018-12-03:
A bug was introduced in the 3.5.0 firmware release related to the M601 gcode command when printing from an SD card.  When this command is executed it is taking the print's current location on the sd card and rolling backwards equal to the length of data in the command buffer and planner.  This results in a new location to be used upon resume.  This new location is immediately before the M601 command.  This results in the long pause operation being called again after resuming from a long pause.

Also, M601 now implicitly pauses as part of the command, where before you would need to use an M1 command to wait for user input before using the M602 resume command.

The bug appears to come from this commit:
https://github.com/prusa3d/Prusa-Firmware/commit/1585dc66b0f96c63dd65b57644326180761ef3c5

The issue resides in Marilin_main.cpp in the function "stop_and_save_print_to_ram"

I have made a workaround, but this should really have a more proper fix:
https://github.com/TheHammockGuy/Prusa-Firmware/commit/867efed04353a11877742bba19288d3d5bfad35a

Current Behavior:
M601 Turns off nozzle heat, moves to a holding location, and triggers an implicit pause.
User selects Resume from LCD menu and it re-executes the above M601 command resulting in a loop

Expected(previous) behavior:
M601 Turns off nozzle heat, moves to a holding location, and then continues executing gcode to allow for
M1 to wait for user input
M602 to reposition back to original location, reheat, and continue printing

Behavior with my workaround:
M601  Turns off nozzle heat, moves to a holding location, triggers an implicit pause and sets a flag to ignore further M601 commands until after M602 is called
User selects Resume from the LCD menu and it skips the above M601 due to the flag setting
M602 Resets the flag allowing M601 to be executed again in the future

TheHammockGuy, 2018-12-03:
I have also created a pull request in case you want to use my workaround until a better fix can be scheduled:

Pull request https://github.com/prusa3d/Prusa-Firmware/pull/1379

PavelSindler, 2018-12-04:
Thank you for reporting this bug.

thenickdude, 2019-01-18:
Looks like this got fixed in 3.5.1 by 16e5fb50ddc72a294c7eb173622d0ec2961210cc

kubiani, 2019-01-18:
> Looks like this got fixed in 3.5.1 by [16e5fb5](https://github.com/prusa3d/Prusa-Firmware/commit/16e5fb50ddc72a294c7eb173622d0ec2961210cc)

I'm still seeing this issue. Currently running v3.5.1-1778 on a Mk2.5. I normally print via USB from an attached OctoPi but I've also confirmed the infinite loop behavior still occurs via an SD print. 

The only apparent difference with the USB print is it seems like it doesn't respect the new "wait for user input" as part of M601 command; on the USB print it tries to auto resume whereas the SD print requires me to select "resume print" from the menu.

Grismo, 2019-01-19:
The specific code in marlin_main.cpp which shall cure the issue is at least not there in the source code for version 3.5.1. So I would assume that it might be a problem for MK3 as well.

Grismo, 2019-03-07:
The M601 bug seems still to be there, although maybe in another form. I just tested M601 in a small test. When M601 is called, printer moves to pause position, hotend is switched of, heatbed is still on. LCD shows message "print paused". But how to resume the print? If I remember correctly there has been a menu item like "resume print" when M601 caused an endless repetition. An combination of M601 followed by M1 doesn't work as well. So now we are caught in an endless pause instead of an endless pause loop.
By the way - I tested it on an MK2.5 with MMU2 with FW 3.6.0.

thenickdude, 2019-05-12:
Confirmed that M601 still doesn't work in 3.7.1-RC1 for the MK3S. M601 results in the print pausing, and "print paused" is displayed on the screen. Selecting Resume Print from the menu results in the printhead returning to the print surface, and making one quick lateral movement, then the print pauses again and the head is retracted - an infinite loop that can't be escaped from.

McSlow, 2019-05-29:
Doesn't work with 3.71 release either. Still infinite pause loop.

F0x06, 2019-07-01:
Still not fixed since 3 Dec 2018 ? This ruined one of my prints...

F0x06, 2019-07-01:
@TheHammockGuy Thanks for the "dirty" fix, patched my v3.7.1 with the same logic, works perfectly until a better fix from Prusa 👍

TheHammockGuy, 2019-07-01:
It may be helpful if people still experiencing this but post the couple lines of gcode before and after your use of M601 to be sure of the exact conditions that you are using to reproduce the bug.  Even if commit [16e5fb5](https://github.com/prusa3d/Prusa-Firmware/commit/16e5fb50ddc72a294c7eb173622d0ec2961210cc) fixed the infinite loop issue, the behavior and use of M601 would still have changed slightly.

As in my original post, you would previously have needed to do:

M601 to initiate pause
M1 to wait for user input
M602 to resume print

With the change to the behavior of M601 you no longer need to follow up with M1 and M602.  It would be good to confirm that the people still experiencing issues are using only M601 and not following up with M1 and M602.

This data may help Prusa engineers release a less hacky fix sooner.

F0x06, 2019-07-01:
> It may be helpful if people still experiencing this but post the couple lines of gcode before and after your use of M601 to be sure of the exact conditions that you are using to reproduce the bug. [...]

I was printing a glasses stand, with some pennies insert at layer 2.6
```
.....
G1 X109.081 Y79.797 E0.19944
G1 X109.473 Y79.864 E0.01343
G1 X110.593 Y80.169 E0.03931
G1 X111.679 Y80.581 E0.03931
G1 X112.720 Y81.096 E0.03931
G1 X113.706 Y81.708 E0.03931
G1 X114.660 Y82.443 E0.04075
G1 X119.864 Y77.239 E0.24912
G1 X121.462 Y77.414 E0.05441
G1 X123.230 Y77.530 E0.05997
G1 X125.001 Y77.568 E0.05997
G1 X126.773 Y77.530 E0.05997
G1 X127.870 Y77.458 E0.03721
G1 X117.931 Y87.396 E0.47574
M204 S1000
M601
;BEFORE_LAYER_CHANGE
;2.6

G1 E-0.04000 F1800.00000
G1 F5760
G1 X118.746 Y86.581 E-0.34200
G1 E-0.01800 F1800.00000
G1 Z2.800 F7200.000
;AFTER_LAYER_CHANGE
;2.6
G1 X99.589 Y68.539
G1 Z2.600
G1 E0.40000 F1800.00000
M204 S800
G1 F4800
G1 X100.053 Y68.539 E0.01573
......
```

The printer paused correctly and parked, then i selected "Resume" from LCD, printer heated-up again, resumed the print, then instantly paused and parked again when the nozzle touched the part, tried 3 times and same behavior.

TheHammockGuy, 2019-07-01:
@PavelSindler @XPila 
The issue seems to be that the stop_and_save_print_to_ram function which is called by lcd_pause_print uses sdpos_atomic to determine the current position on the SD card for saving print progress.  The variable sdpos_atomic is not impacted by the cmdqueue_pop_front function in the commit that was intended to fix this bug.  So even though that pop command is being executed the pause function is still using the same SD card position to resume the print as it was before the pop was requested.

robscar, 2019-11-08:
still seeing the exact same issue in 3.8.1. Any chance to fix this?

kenour, 2019-11-19:
Oh it's not just me! I've been trying to add a pause at a specific layer to pour epoxy resin into my print, but it gets stuck in a loop. I've been using PrusaSlicer and would go to the layer I wanted, use the GUI to insert a filament change, then just change the M600 for M601. But it wouldn't restart after the pause.

Filament change works, but I really want a long pause as I need the extruder to bugger off out the way and the model to be moved forward to pour the resin in.

wavexx, 2019-11-19:
See PR #2274

Mito57, 2019-12-22:
This bug has not been fixed. I just tried the new "add pause" feature in the last Prusa slicer and I caused the continuous loop. I had to stop the print and insert a different code manually into the gcode.  
I used this:
https://forum.prusaprinters.org/forum/original-prusa-i3-mk2-s-others-archive/pause-print-in-gcode/#post-125748
Although not as good as the M601 should be, it works.

wavexx, 2019-12-23:
Are you printing via usb or via SD?

Mito57, 2019-12-23:
Via SD

wavexx, 2019-12-23:
There are still some PRs which deal with pause/resume issues (#2352, #2353) that got discovered later which are not yet contained in an official release, and also a few other ones that still need merging (#2368, #2382 and #2383) to cover some remaining issues.

With these PRs merged I can use M601 correctly both from the SD and via USB (just tested with a small test cube and with the new "add pause"/M601 added by slic3r).

I could build a test firmware if you're willing to help testing it :) (in this case, please say which model do you have).

Mito57, 2019-12-24:
I have never done any firmware modification or compiling but I can
definitely test if if you do it. My printer is a MK3S and I'm running the
3.8.1-2869 firmware.

pavelchuchma, 2019-12-26:
The issue reproduced using prusa3d_fw_3_8_1_MK3. Printing from SD card. 
I tried to resume ~5 times, no success. Successfully resumed after unload and load the filament.
Attaching my model.
[bok01-zaklad_0.15mm_PETG_MK3_59m.zip](https://github.com/prusa3d/Prusa-Firmware/files/4003602/bok01-zaklad_0.15mm_PETG_MK3_59m.zip)

Mito57, 2019-12-27:
That's exactly what happened to me but I did not think on unloading/loading filament, but then the workaround would be to insert a color change instead of a pause. A little more work but it will stop and resume the print successfully. I've done it before the context menu was added to the slicer.
Prusa should just fix the bug in the firmware or remove the option in the context menu. I used the option to insert this gcode (using the context menu) and it worked just fine. 

G1 X10.000 Y200.000 E0; parking position 
M1; user stop
M105; return to current temp

pavelchuchma, 2019-12-27:
I believe that "Průšovi kluci šikovní" will fix this forgotten, simply reproducible, bug very soon :-)

wavexx, 2020-01-02:
I prepared two builds for those that would like to test the pause/resume functions.

* For the  MK3: [FW381-Build2869-1_75mm_MK3-EINSy10a-E3Dv6full-EN_ONLY.zip](https://github.com/prusa3d/Prusa-Firmware/files/4016736/FW381-Build2869-1_75mm_MK3-EINSy10a-E3Dv6full-EN_ONLY.zip)
* For the MK3S: [FW381-Build2869-1_75mm_MK3S-EINSy10a-E3Dv6full-EN_ONLY.zip](https://github.com/prusa3d/Prusa-Firmware/files/4016739/FW381-Build2869-1_75mm_MK3S-EINSy10a-E3Dv6full-EN_ONLY.zip)

These build on top of the current master 955c88cf688a0456868304b5c1fc4ab34a9abddf and additionally include PRs #2382 #2383 #2394 #2393 #2390 #2368 #2395 in single-language (English only).

This is unofficial/testing stuff, so you know the risks. I tried all combinations I could think of (M601 via USB or SD, M600, change of filament (M600) via tune during a pause triggered via M601) but I'd love if somebody should give it more testing.

Mito57, 2020-01-04:
I tested the pause function with this firmware and appears to be working
fine now. I used the "insert pause" from the Prusa slicer beta using the
context menu.

I'm not really sure how it's supposed to work though. I had to go to the
"resume print" option on the printer's menu to keep going. I thought that
is should have displayed a custom message and only pressing the button was
needed to resume?

wavexx, 2020-01-04:
On Sat, Jan 04 2020, Mito57 wrote:
> I'm not really sure how it's supposed to work though. I had to go to the
> "resume print" option on the printer's menu to keep going.

That's how it's supposed to work using the M601 instruction inserted by
PrusaSlicer. It's a generic pause function: it does exactly the same as
if you paused the print via the LCD.

There's a custom message too (using M117), but it's displayed for a
fraction of a second only before being replaced by "Print paused".

This needs to be improved.

> I thought that is should have displayed a custom message and only
> pressing the button was needed to resume?

There's a different gcode for this, M0, which should prompt for the user.
I didn't test this one yet.

I did use M0 with stock marlin, but I still recommend to use M601
instead. With a regular pause you can use the settings menu to move the
extruder higher if you need, and then also resume the print. Very
useful!

clintonefisher, 2020-02-22:
Yes!  Using 3.9.0-RC1, I was able to pause the print (a 3D BB Maze), install a captive BB, and successfully resume the print.  Fantastic!  Thanks everyone!

kingigi, 2020-03-22:
✅ : The Prusa Firmware 3.9.0-RC1-3272 solved the pause-resume-loop problem for me too.
I used PrusaSlicer 2.2.0 to slice the model.

bivaterl, 2020-04-09:
I was having the infinite loop issue (not infinite; I was able to do it over and over and eventually it resumed). However, due to this thread, I installed 3.9.0 RC1 (whatever is current as of 4/9/2020) and the loop issue was resolved!
Penny slugs to give weight to my plastic coins achievement unlocked!

cnlson, 2020-04-21:
I just had this happen when using a pause at height from 2.2.2 prusa slicer.
Lost that print and re sliced with filament change and that worked to drop in a nut.

jessearmstrong, 2020-05-04:
I'm also having this loop on my MK2.5s

leptun, 2020-05-04:
@jessearmstrong what fw version?

jessearmstrong, 2020-05-04:
I am running 3.8.1 on my MK2.5s

wmajst, 2020-05-24:
I use such workaround for tihs:
use M601, when paused take off sd, replace M601 with M602, insert card, use LCD to resume print

WRidder, 2020-05-25:
I encountered this issue as well, experiencing an infinite pause/resume loop upon inserting a pause. Upgrading to 3.9.0 fixed the issue for me.

3d-gussner, 2020-07-13:
@TheHammockGuy As we fixed this issue it would be great if you could close this issue.
[stale][fixed]

JohnnyDeer, 2020-08-12:
I'm closing this issue as stale, thanks for understanding.

closed 2020-08-12 by JohnnyDeer