prusa3d/Prusa-Firmware #1380: Error Compiling PRINTER_TYPE

Sjaakwortel, 2018-12-04:
Followed the instructions from the readme to compile marlin for mk2.5, compiling gives error : 

" exit status 1 'PRINTER_TYPE' was not declared in this scope" 
This is connected to M45, but i cant seem to solve this, tried different versions of the arduino enviroment.

timmmccoy, 2019-01-03:
I was able to solve this by editing the Configuration_prusa.h from the Firmware folder _instead_ of copying over the version from the variants folder.

With a code diff tool you can compare the variant file you chose against the "default" Configuration_prusa.h and figure out which edits you need to make.

Hope this helps.

3d-gussner, 2019-10-08:
@Sjaakwortel Long time no update here. Can you please close this issue?

closed 2022-12-29 by wavexx