prusa3d/Prusa-Firmware #1384: [3.5.0] MK3. X-position lost on unpause

nojan1, 2018-12-06:
After having performed a pause during the print it seems the X-position was somehow lost or corrupted. This caused all the subsequent extrusions to be made about 4 cm to the right of their correct locations, ruining the print.

Maybe the printer should perform a re-home after unpause to make sure it knows where it is?

Pathogenius, 2018-12-06:
3.5.0-RC1 (1688), I recently did a print which required a dozen or so M600 type filament changes, which I assume involves lots of 'pauses', print came out fine, no loss of position.

PavelSindler, 2018-12-06:
Interesting... We will check again if steppers are not switched-off during pause.

lohring, 2018-12-21:
I have a similar issue with my Mk3.  When printing objects that are close to the right edge the X axis looses position and moves several mm to the right.  This usually happens in the middle of the print and results in a crash.  The XYZ calibration must be rerun or subsequent prints will be offset.  Is this a circuit board problem?  I've reflashed and recalibrated the printer several times.  The printer ran perfectly for several months before this happened.

Georsch, 2019-01-24:
Nach Firmware-Update das gleiche Problem mit der X-Achse. Bei einem Crash oder Neustart des Druckers nimmt die X-Achse die letze Position als 0-Punkt an. Abhilfe schafft nur die X-Achse per Hand auf die 0-Position zu Fahren. Entweder nach Motor aus oder Nertz aus. Danach fährt die X-Achse auch wieder auf null. Mit der XYZ Kalibrierung funktioniert auch keine Referenzfahrt der X-Achse gegen null.

RichardHerold, 2019-09-11:
I'm having the same issue as Georsch. Since 3.6 every time i have to move it manually by hand. There is definitely a major regression here that should be looked at

closed 2023-09-14 by github-actions[bot]