prusa3d/Prusa-Firmware #1388: Communication issues after flashing firmware

ChrisVahi, 2018-12-07:
After uploading my build of FW 3.5.0 with changes to extruder position and build volume my printer would stall on the startup screen or show the info screen but would timeout while connecting to Octoprint(pi3 B+) and Repetier(Windows 10). I had my MMU power cable disconnected during flashing after getting frustrated with inadequate power during homing, not sure if this has something to do with this.

The firmware updater (I had been using the Firmware Updater plugin on Octoprint, which has not failed me yet) showed a successful flash, however, after that Octoprint could not connect to the printer. Upon clicking "Connect" the printer would do the normal reset and start to connect, but would then time out after about 30 seconds. The serial monitor showed the output pasted below. I tried flashing again with similar results, only the LCD started flashing on the initial title screen and would not finish booting up. I tried a factory reset but it seemed like the boot process never reached that checkpoint. I then flashed the stock version of 3.5.0 and got an error message "External SPI flash W25X20CL not responding." After researching this fault it seems to be related to the daughter board released on earlier models. My einsy never came with one. 

I did manage to get my previous working version of 3.4.1(debug build) to upload and get past the SPI flash screen (though that screen does appear on boot). I was also able to make my corrections and reflash 3.4.1 and it worked for one print. I should also mention that I did manage to successfully flash 3.5.0 two or three times today before this problem started. I don't think it has to do with my computer or the version of Arduino that I'm using to compile. However I did have difficulty compiling after this started. I kept getting errors from the d_codes sketches around missing the declaration of the routine LOG() and the variable axis_steps_per_unit(). Eventually I got around this by uncommenting sections of code, which allowed me to complete compiling, but did not fix the printer. I have seen a variety of errors, all seeming to do with communication. My SD card is now unreadable by the printer, though perfect in Windows. I have also seen several "Rx buffer full" errors pop up. 

Here's my serial log:

2018-12-03 10:34:58,728 - Recv: echo: 3.5.0-1749
2018-12-03 10:34:58,732 - Recv: echo: Last Updated: Nov 30 2018 12:01:48 | Author: (none, default config)
2018-12-03 10:34:58,736 - Recv: Compiled: Nov 30 2018
2018-12-03 10:34:58,740 - Recv: echo: Free Memory: 2084  PlannerBufferBytes: 1392
2018-12-03 10:34:58,744 - Recv: echo:Stored settings retrieved
2018-12-03 10:34:58,746 - Recv: adc_init
2018-12-03 10:34:59,079 - Recv: CrashDetect DISABLED
2018-12-03 10:34:59,424 - Recv: PAT9125_RES_X=0
2018-12-03 10:34:59,429 - Recv: PAT9125_RES_Y=240
2018-12-03 10:34:59,431 - Recv: PAT9125_init:1
2018-12-03 10:34:59,433 - Recv: FSensor DISABLED
2018-12-03 10:34:59,436 - Recv: 
2018-12-03 10:34:59,518 - Recv: echo:SD card ok
2018-12-03 10:35:02,523 - Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-12-03 10:35:02,526 - Send: M105
2018-12-03 10:35:02,537 - Recv: ok T:228.9 /0.0 B:85.1 /0.0 T0:228.9 /0.0 @:0 B@:0 P:26.7 A:39.0
2018-12-03 10:35:03,804 - Send: M105
2018-12-03 10:35:03,815 - Recv: ok T:228.0 /0.0 B:85.0 /0.0 T0:228.0 /0.0 @:0 B@:0 P:26.7 A:38.8
2018-12-03 10:35:08,805 - Send: M105
2018-12-03 10:35:08,816 - Recv: ok T:225.8 /230.0 B:84.7 /85.0 T0:225.8 /230.0 @:45 B@:0 P:26.7 A:38.8
2018-12-03 10:35:13,806 - Send: M105
2018-12-03 10:35:13,818 - Recv: ok T:223.5 /230.0 B:84.1 /85.0 T0:223.5 /230.0 @:105 B@:0 P:26.6 A:38.5
2018-12-03 10:35:15,395 - Recv: MMU version valid
2018-12-03 10:35:15,398 - Recv: MMU - ENABLED
2018-12-03 10:35:15,807 - Send: M105
2018-12-03 10:35:15,816 - Recv: ok T:224.1 /230.0 B:83.9 /85.0 T0:224.1 /230.0 @:100 B@:0 P:26.9 A:38.5
2018-12-03 10:35:17,809 - Send: M105
2018-12-03 10:35:17,819 - Recv: ok T:224.9 /230.0 B:83.7 /85.0 T0:224.9 /230.0 @:95 B@:0 P:26.7 A:38.4
2018-12-03 10:35:19,810 - Send: M105
2018-12-03 10:35:19,823 - Recv: ok T:225.9 /230.0 B:83.4 /85.0 T0:225.9 /230.0 @:85 B@:69 P:26.6 A:38.7
2018-12-03 10:35:21,811 - Send: M105
2018-12-03 10:35:21,822 - Recv: ok T:227.9 /230.0 B:83.3 /85.0 T0:227.9 /230.0 @:59 B@:127 P:26.8 A:38.3
2018-12-03 10:35:23,813 - Send: M105
2018-12-03 10:35:23,825 - Recv: ok T:229.5 /230.0 B:83.0 /85.0 T0:229.5 /230.0 @:39 B@:127 P:26.7 A:38.3
2018-12-03 10:35:25,814 - Send: M105
2018-12-03 10:35:25,827 - Recv: ok T:230.6 /230.0 B:82.8 /85.0 T0:230.6 /230.0 @:27 B@:127 P:26.6 A:38.3
2018-12-03 10:35:27,818 - Send: M105
2018-12-03 10:35:27,830 - Recv: ok T:232.1 /230.0 B:82.8 /85.0 T0:232.1 /230.0 @:7 B@:127 P:26.6 A:38.5
2018-12-03 10:35:29,819 - Send: M105
2018-12-03 10:35:29,829 - Recv: ok T:232.4 /230.0 B:82.8 /85.0 T0:232.4 /230.0 @:7 B@:127 P:26.7 A:38.6
2018-12-03 10:35:31,820 - Send: M105
2018-12-03 10:35:31,833 - Recv: ok T:231.8 /230.0 B:82.9 /85.0 T0:231.8 /230.0 @:21 B@:127 P:26.9 A:38.5
2018-12-03 10:35:33,822 - Send: M105
2018-12-03 10:35:33,832 - Recv: ok T:231.4 /230.0 B:83.1 /85.0 T0:231.4 /230.0 @:29 B@:127 P:26.8 A:38.6
2018-12-03 10:35:35,823 - Send: M105
2018-12-03 10:35:35,835 - Recv: ok T:230.7 /230.0 B:83.2 /85.0 T0:230.7 /230.0 @:38 B@:127 P:26.8 A:39.1
2018-12-03 10:35:37,703 - Recv: USART2 rx Full!!!
2018-12-03 10:35:37,825 - Send: M105
2018-12-03 10:35:37,837 - Recv: ok T:229.8 /230.0 B:83.5 /85.0 T0:229.8 /230.0 @:52 B@:101 P:26.8 A:39.0
2018-12-03 10:35:39,827 - Send: M105
2018-12-03 10:35:39,836 - Recv: ok T:229.1 /230.0 B:83.6 /85.0 T0:229.1 /230.0 @:65 B@:90 P:26.9 A:39.3
2018-12-03 10:35:41,828 - Send: M105
2018-12-03 10:35:41,840 - Recv: ok T:229.2 /230.0 B:83.9 /85.0 T0:229.2 /230.0 @:60 B@:67 P:26.8 A:39.0
2018-12-03 10:35:43,831 - Send: M105
2018-12-03 10:35:43,843 - Recv: ok T:229.4 /230.0 B:84.1 /85.0 T0:229.4 /230.0 @:57 B@:52 P:26.8 A:39.1
2018-12-03 10:35:45,835 - Send: M105
2018-12-03 10:35:45,870 - Recv: ok T:229.8 /230.0 B:84.3 /85.0 T0:229.8 /230.0 @:50 B@:48 P:26.8 A:39.2
2018-12-03 10:35:47,836 - Send: M105
2018-12-03 10:35:47,849 - Recv: ok T:230.7 /230.0 B:84.5 /85.0 T0:230.7 /230.0 @:32 B@:45 P:26.8 A:39.3
2018-12-03 10:35:49,840 - Send: M105
2018-12-03 10:35:50,200 - Recv: echo:busy: processing
2018-12-03 10:35:52,215 - Recv: echo:busy: processing
[... 21 lines cut ...]
2018-12-03 10:36:36,547 - Recv: echo:busy: processing
2018-12-03 10:36:38,563 - Recv: echo:busy: processing
2018-12-03 10:36:40,578 - Recv: echo:busy: processing
2018-12-03 10:36:42,590 - Recv: echo:busy: processing
2018-12-03 10:36:44,605 - Recv: echo:busy: processing
2018-12-03 10:36:46,621 - Recv: echo:busy: processing
2018-12-03 10:36:48,636 - Recv: echo:busy: processing
2018-12-03 10:36:50,654 - Recv: echo:busy: processing
2018-12-03 10:36:52,666 - Recv: echo:busy: processing
2018-12-03 10:36:54,681 - Recv: echo:busy: processing
2018-12-03 10:36:56,801 - Recv: echo:busy: processing
2018-12-03 10:36:58,712 - Recv: echo:busy: processing
2018-12-03 10:37:00,728 - Recv: echo:busy: processing
2018-12-03 10:37:02,743 - Recv: echo:busy: processing
2018-12-03 10:37:04,757 - Recv: echo:busy: processing
2018-12-03 10:37:06,769 - Recv: echo:busy: processing
2018-12-03 10:37:08,784 - Recv: echo:busy: processing
2018-12-03 10:37:10,801 - Recv: echo:busy: processing
2018-12-03 10:37:13,837 - Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-12-03 10:37:13,845 - Send: M105
2018-12-03 10:37:16,853 - Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-12-03 10:37:16,856 - Send: M105
2018-12-03 10:37:21,884 - No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-12-03 10:37:21,896 - Changing monitoring state from "Operational" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
2018-12-03 10:37:21,916 - Connection closed, closing down monitor

inquisitiveKangaroo, 2018-12-12:
Hi Chris,
So it was eventually successful by Slic3r PE?
This shall be related to the language packages which were comprimated into single fw.
The flashing is supported only by Slic3r PE furthermore.

ChrisVahi, 2018-12-12:
> No, I just included the slic3r log for reference. The electronics are
> still unusable. After flashing a debug build of 3.4.1 through the Octoprint
> plugin right after the trouble started I got one print before it finally
> bricked. I am able to flash  "successfully" every time but all other
> communication fails including serial and SD card. I am not able to print at
> this time. I will try to comment out the 2Mb flash chip and see if I can
> boot to eliminate possible faults.
>
> Please advise,
> Chris
>
> On Wed, Dec 12, 2018, 8:07 AM inquisitiveKangaroo <

What I would really like to know is if you have a specific bootloader I can
try to start from a clean slate. I've seen others on the forums had this
issue pop up and reflashing seemed to fix it but my problem seems deeper.

inquisitiveKangaroo, 2018-12-12:
Right now the i3 MK3 fw with language packages **must** be flashed through Slic3r PE (4 months old at maximum). The only possibility to flash it through the Octoprint is by adding a modified _avrdude_ and call it twice (once for the fw and once for the language packages).
I would be glad if you let us know about your progress. Frankly, I have not use anything but slicer since the language packages went out (always worked well) but it is always good to know about alternatives. 
Wish you good luck Chris and I hope I helped you atleast a little bit:)

inquisitiveKangaroo, 2018-12-12:
Do you know what type of Einsy you have? You can find it in the bottom right corner of the board.

[image]

If you have 1.0a you require a special reduction which you put in between the lcd-cabel and the board.
(if you have this version without the reduction, please contact the customer support)
If you have 1.1a it should be working without any additional reduction.

closed 2023-09-14 by github-actions[bot]