prusa3d/Prusa-Firmware #1390: Compiled firmware not working

F0x06, 2018-12-08:
Version. 9adf441d5ced473996fd076ee13b914e6f05bf62
Printer: MK3

Hi, when i compile the firmware using `build.sh` for my MK3, this is the log i get, i think something is wrong with languages, when i flash the firmware.hex my printer display random chars really fast and bootloops
```
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 219490 bytes (85%) of program storage space. Maximum is 258048 bytes.
Global variables use 6127 bytes of dynamic memory.
lang-build.sh started
lang=en
count=331
offs=678
size=9610
lang=cz
count=331
offs=678
size=9460
lang=de
count=331
offs=678
size=10567
lang=es
count=331
offs=678
size=10418
lang=fr
count=331
offs=678
size=10812
lang=it
count=331
offs=678
size=10462
lang=pl
count=331
offs=678
size=9873
lang-build.sh finished with success
config.sh started
 Arduino main folder: OK
 Arduino builder: OK
 AVR gcc tools:
   objcopy OK
   objdump OK
 Output folder: OK
 Objects folder: OK
 Generated elf file: OK
 Generated hex file: OK
config.sh finished with success
postbuild.sh started
 checking files:
  folder  OK
  elf     OK
  objects OK
 running progmem.sh...OK
 running textaddr.sh...OK
 checking textaddr.txt...OK
 extracting binary...OK
 updating binary:
  primary language ids...OK
  primary language signature...OK
 converting to hex...OK
  secondary language data...Updating languages:
 Czech  : NG!
 German : NG!
 Italian: NG!
 Spanish: NG!
 French : NG!
 Polish : NG!
 converting to hex...OK

postbuild.sh finished with success
```

Thanks in advance !

Panayiotis-git, 2018-12-08:
Same here. Compile and upload with arduino as usual.

maximlevitsky, 2018-12-09:
Note that this does work for me.

Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 244228 bytes (94%) of program storage space. Maximum is 258048 bytes.
Global variables use 6127 bytes of dynamic memory.
lang-build.sh started
lang=en
count=331
offs=678
size=9610
lang=cz
count=331
offs=678
size=9460
lang=de
count=331
offs=678
size=10567
lang=es
count=331
offs=678
size=10418
lang=fr
count=331
offs=678
size=10812
awk: cmd. line:1: (FILENAME=- FNR=170) warning: Invalid multibyte data detected. There may be a mismatch between your data and your locale.
lang=it
count=331
offs=678
size=10462
lang=pl
count=331
offs=678
size=9873
lang-build.sh finished with success
config.sh started
 Arduino main folder: OK
 Arduino builder: OK
 AVR gcc tools:
   objcopy OK
   objdump OK
 Output folder: OK
 Objects folder: OK
 Generated elf file: OK
 Generated hex file: OK
config.sh finished with success
postbuild.sh started
 checking files:
  folder  OK
  elf     OK
  objects OK
 running progmem.sh...OK
 running textaddr.sh...OK
 checking textaddr.txt...OK
 extracting binary...OK
 updating binary:
  primary language ids...OK
  primary language signature...OK
 converting to hex...OK
  secondary language data...Updating languages:
 Czech  : OK
 German : OK
 Italian: OK
 Spanish: OK
 French : OK
 Polish : OK
 converting to hex...OK

postbuild.sh finished with success

Stock fedora 28 here.

F0x06, 2018-12-09:
Tried on Ubuntu and it works... Investigated the problem a bit, it seems that the `objdump` command used to generate symbols that allows lang scripts to find the addresses to insert languages cant find the section, im running ArchLinux

Panayiotis-git, 2018-12-13:
The compile instructions have been updated:

2.c. In file "Firmware/config.h" set LANG_MODE to 0.

With this I managed to build and upload the firmware without boot loops.

3d-gussner, 2018-12-21:
Hi,  I use Windows and found a way to compile the Prusa MK3 and MK2.5 firmware. Which should also work on all Linux versions.
I use Arduino IDE 1.6.9 and 1.8.5 in portable mode exclusively for the Prusa environment.
I will try to update my https://github.com/3d-gussner/Prusa-Firmware/wiki/How-to:-Build-Prusa-Firmware asap.
If you cant wait just checkout https://github.com/mkbel/PF-build-env/commit/395c94bdd63c5519ad7b5cebd8463ba809f26518 these changes to avr6.xn made the difference between
```
 Czech  : NG!
 German : NG!
 Italian: NG!
 Spanish: NG!
 French : NG!
 Polish : NG!
```
and 
```
Czech : OK
German : OK
Italian: OK
Spanish: OK
French : OK
Polish : OK
```

@mkbel How do you generate the initial lang_en.txt file? And how does the lang_add.txt has to be to add new missing messages?
There are also some not_used messages, some of them are MK2.5 or MK3 specific but other aren't translated even they are used in both versions like:
```
Change extruder
Extruder 1
Extruder 2
Extruder 3
Extruder 4
Fil. sensor response is poor, disable it?
Improving bed calibration point
Insert filament into extruder 1. Click when done.
Insert filament into extruder 2. Click when done.
Insert filament into extruder 3. Click when done.
Insert filament into extruder 4. Click when done.
Iteration
No 
Unload all
XYZ calibration compromised. Left front calibration point not reachable.
XYZ calibration failed. Left front calibration point not reachable.

```

How do i get these added so i can translate them correctly.

Which software do you use for the `.pot` and `.po` files? Something like https://poedit.net and if so which version?

And final question @mkbel @PavelSindler 
as the ../lang/config.sh shows `export ARDUINO=C:/arduino-1.8.5` and the PF-build-env revision is also 1.8.5
are we finally allowed to use Arduino IDE 1.8.5 instead of the old unsupported Arduino IDE 1.6.9?

3d-gussner, 2019-01-06:
Hi everybody. It took me some time but here a Windows build environment you can try if you want
https://github.com/3d-gussner/Prusa-Firmware/tree/MK3-win_build

mkbel, 2019-01-08:
@XPila Can you please answer that?
How do you generate the initial lang_en.txt file? And how does the lang_add.txt has to be to add new missing messages?
There are also some not_used messages, some of them are MK2.5 or MK3 specific but other aren't translated even they are used in both versions like:

Change extruder
Extruder 1
Extruder 2
Extruder 3
Extruder 4
Fil. sensor response is poor, disable it?
Improving bed calibration point
Insert filament into extruder 1. Click when done.
Insert filament into extruder 2. Click when done.
Insert filament into extruder 3. Click when done.
Insert filament into extruder 4. Click when done.
Iteration
No 
Unload all
XYZ calibration compromised. Left front calibration point not reachable.
XYZ calibration failed. Left front calibration point not reachable.

How do i get these added so i can translate them correctly.

Which software do you use for the .pot and .po files? Something like https://poedit.net and if so which version?

3d-gussner, 2019-01-08:
@mkbel Glad to see Prusa changed to IDE 1.8.5 and thanks for asking @XPila my questions again.

3d-gussner, 2019-02-16:
Checkout https://github.com/3d-gussner/Prusa-Firmware/wiki/How-to:-Build-Prusa-Firmware

3d-gussner, 2019-02-21:
https://github.com/prusa3d/Prusa-Firmware/pull/1543 is pulled and should make compiling easier.

3d-gussner, 2019-10-08:
@F0x06 Long time no update. Can you please close this issue?

closed 2019-10-08 by F0x06