prusa3d/Prusa-Firmware #1391: 3.5.0 Extruder Skipping

bergerda, 2018-12-08:
Hello,
I am sure you have heard of this issue time and time again. I have an issue with the MK3 skipping steps after this update. I checked my hardware for the common issues, gears, extruder jam, too cold/hot of a temperature... etc. However, when I feel the extruder stepper motor, it is very very warm to the touch.
Let me know if this gets figured out.
Thanks!

MimbaMonkeyHouse, 2018-12-10:
Same here. Downgrading after 3 days of misery!

inquisitiveKangaroo, 2018-12-12:
Hi,
Thank you for report. We will check it out.)

Prusa-Support, 2022-05-10:
This thread could be related to issue #765.
Also, this thread has been silent for years.
For these reasons, this issue will be closed.

Please check the previous issue.
Comment below or feel free to open a new issue if you feel this issue deserves a separate discussion.

Michele Moramarco
Prusa Research

closed 2022-05-10 by Prusa-Support