prusa3d/Prusa-Firmware #1392: Filament auto loading issue

maximlevitsky, 2018-12-09:
Basically the following always results in printer going crazy:

1. Ensure that no filament is loaded and filament loading is enabled, printer is preheated.
2. Insert the filament such as you see that filament loading is started but as soon as filament sensor starts the load pull it out and then it again (even better do this few times).
3. Now let the gears grab the filament.

4. Wait till you see the message about the correct color, and answer yes, or let it load more (doesn't matter).

5. Instead of beeing done, printer moves the print head X position to the rightmost, asks you to unload the fillament, what looks like a filament swap.
 
6. If you proceed with the instructions (unload the filament, load it), printer will eventually start printing a random file from the SD card. The menu will not contain 'stop print' option.

This happens on 3.5.0.

What I think is happening, is that if you remove the filament during loading, this triggers the filament swap since during loading the printer sort of 'printing' (I know from source that menu code sends hardcoded G-codes to the planner for most of the tasks like fillament load).

The fix would be to disable the fillament auto-loading trigger during the any kinds of user started operations (like axis moving, fillament loading, etc ) when printer is not printing.

On some occasions I was able to make the printer go crazy in other way:
Once it showed that it loading the filament but the gear was rotating backwards for lot of time till I reset the printer/

Note the other report about filament load (#1366) very similar to what I get.
But in this case, fillament autoloading is disabled, since in the report it was stated that 'filament load' menu was chosen.

VentyCZ, 2018-12-09:
This also happened to me the other day.

The strange thing was, that no SD card was in the printer, but it still tried to print a StarGate REPLICATOR model!

Coincidence ? I think NOT!

PS: I might record it on camera, if I get a chance.

VentyCZ, 2018-12-18:
Oh, did not mention this:
It also happens with manual filament loading.
And sometimes the gears are moving in the opposite direction while loading (unload procedure instead of loading one)

cnlson, 2018-12-29:
i don't autoload the filament and i did notice this with manual filament loading. if you hit NO so it would purge more filament it would randomly start slowly turning the gears backwards and unloading the filament.

leptun, 2018-12-29:
> What I think is happening, is that if you remove the filament during loading, this triggers the filament swap since during loading the printer sort of 'printing' (I know from source that menu code sends hardcoded G-codes to the planner for most of the tasks like filament load).
> 
> The fix would be to disable the filament auto-loading trigger during the any kinds of user started operations (like axis moving, filament loading, etc ) when printer is not printing.
> 
> On some occasions I was able to make the printer go crazy in other way:
> Once it showed that it loading the filament but the gear was rotating backwards for lot of time till I reset the printer/

I did some digging and found out that the filament runout can be triggered while the filament is loading. That's why it is going backwards. The filament runout function then continues printing what is probably the first item in the sorted list. (I think).

> Note the other report about filament load (#1366) very similar to what I get.
> But in this case, filament auto loading is disabled, since in the report it was stated that 'filament load' menu was chosen.

It happens regardless of mode since it detects the runout.

leptun, 2018-12-30:
I've been able to fix the issue by skipping the error checking in the ISR during filament loading.
I don't know if this affects mmu as I heard that it is disabled there.

Also I don't know how to do a pull request sooo...

in fsensor.cpp

replace "if (st_cnt != 0)" with "if (st_cnt != 0 && !loading_flag)"

closed 2023-09-14 by github-actions[bot]