prusa3d/Prusa-Firmware #1393: Allow remapping of the tool/extruder table

mgg4, 2018-12-09:
Currently the "T0" command always maps to the first extruder channel, with "T1" selecting the next, and so on.  This means that when a file is sliced and printed, you must be careful to load the correct filament colors in each channel.  One desires to reprint a previously sliced file, only using different colors, rather than swap the filaments into different channels, a method for mapping a tool to an extruder should be provided.

That is, I should be able to map T0 to Extruder 4, and have the MMU select the desired filament from the 4th channel on the MMU.  When the printer is reset would be to map the tools as they are now (T0->E1, T1->E2, etc.).  Multiple tools could be mapped to the same extruder, making both tools print the same color.  

The menu would include a new option for "Map Tools".  When selected, each tool would be listed, along with it's current mapping.  Selecting one of the menu options would open the sub menu to select which filament to map to that tool.  Selecting the filament takes you back to the menu, showing the list including the new selection.  Also on the tool menu, would be a reset to default option.

michalxfanta, 2018-12-10:
Thanks for the idea, its implementation will be included in one of the future releases.

closed 2023-09-14 by github-actions[bot]

autonumous, 2023-10-13:
I know it's closed, but was this remapping feature ever added?   Would be great to have it