prusa3d/Prusa-Firmware #1394: MK3 3.5.0 Randomly stops printing without any reported error

jonnieZG, 2018-12-09:
MK3 with 3.5.0, happened a few times after the fw upgrade that the printer just stops mid print, or at the very beginning during intro-line, without any error being reported on the display. Printing PLA, enclosure in a cold environment, temp inside enclosure 18°C.

Head and bed temperatures remain at the set temperature 215/60°C.

MimbaMonkeyHouse, 2018-12-10:
Same issue here

GWdd, 2018-12-10:
Same here --- in my case printer is working normally at 0.150, and 0.200 layers,  but was doing this "pause" every once in a while when building a 0.100 layer part.  There were overhangs and bridges that may have been involved.  It could be as simple as an oddball Slic3r setting, but that doesn't explain the printer moving along, and just stopping -- no head movement up, sideways, etc. Just extrudes to a point and sits there.

I do have a .gcode from the event.  

[part_pauses_during_print.zip](https://github.com/prusa3d/Prusa-Firmware/files/2663960/part_pauses_during_print.zip)

jonnieZG, 2018-12-11:
>  Just extrudes to a point and sits there. Then a minute later it starts going again. Print finishes as expected.

Please beware that the situation you are describing is **not** the same issue that I reported. In my case the print **does not resume**. In the case that I am reporting, the head sits in a place until I reset the printer.

ChrisVahi, 2018-12-13:
Your g code has a lot of changes to acceleration and jerk settings. Are these your settings or did Prusa as a bunch of noise to the start code? If they are your changes then this may be the problem.

 Acceleration values have to be processed by the firmware before moves can be made. If resulting values exceed the limit for the variable type (eg. >255 for a byte) then the firmware hangs and your print is lost. Try setting your acceleration in eeprom (Google Marlin eeprom acceleration) and reset your print acceleration values in Slic3r to default. I've also seen linear advance (found in filament custom g code tab) cause hangups when values get too high.

GWdd, 2018-12-13:
The code was generated by Slic3r PE,  though I have rearranged the Start-Up gcode segment to reduce ooze during bed level calibration (initial nozzle temp at 175 rather than 215).  And I didn't post-process the gcode. 

If I did anything else, it was only via menus for overhang (slower print speeds) or fill (limiting speed of infill).  I was having issues getting a couple small 0.2mm gaps to print correctly.

jonnieZG, 2018-12-14:
Guys, please don't hijack this thread, since it has nothing to do with the topic in the subject.

GWdd, 2018-12-14:
If you suspect the FW, then down-rev to 3.4 and provide assistance to the developers by demonstrating it is 3.5 that is causing your issue.  Then provide some evidence the problem is unrelated to something in your own gcode - like an example of the problem happening on a common debug print like Benchy..

blgli, 2018-12-28:
I am having the same problem with 3.5.0 on my MK3

I print with ABS and the printer in an enclosure. There's really no rhyme or reason why this happens. Just happened about 3 hours into 8 hour print. Print head just stopped where it was at, heaters for both the bed and hotend were turned off, screen had no errors what so ever on it. It does not happen at the same spot on print, its sporadic where it decides to stop. I've noticed a power cycle of the machine "usually" fixes this until it happens on another print. This has happened to me on numerous occasions after switching to 3.5.0 and I've had enough. Until something can be done to remedy this I will go back to older firmware.

juric57, 2021-05-10:
> MK3 mit 3.5.0, passierte ein paar Mal nach dem fw-Upgrade, dass der Drucker nur mitteldruck stoppt, oder ganz am Anfang während der Intro-Linie, ohne dass ein Fehler auf dem Display gemeldet wurde. Der **Drucker bleibt stecken, bis ein manueller Reset durchgeführt wird.** Drucken von PLA, Gehäuse in einer kalten Umgebung, Temperatur im Inneren Gehäuse 18°C.
> 
> Die Kopf- und Betttemperaturen bleiben bei der eingestellten Temperatur 215/60°C.
> 
> Wenn es sich um ein **temperaturbedingtes Problem (was wahrscheinlich ist)**oder ein anderes Problem handelt, sollte zumindest eine Meldung auf dem Display angezeigt werden.
> 
> ** _BITTE BEACHTEN SIE,_ ** dass dies **nicht** dasselbe ist wie das Problem, wenn ein Drucker während eines Drucks nach dem Zufallsprinzip lange Pausen einlegt, aber nach einer Weile (nach einigen Sekunden oder einer Minute) von selbst wieder aufgenommen wird.

Did you find the reason causing this problem???

juric57, 2021-05-10:
Haben Sie den Grund für dieses Problem gefunden???

amthe, 2023-01-14:
`; stop printing object soap.stl id:0 copy 0
;LAYER_CHANGE
;Z:7.8
;HEIGHT:0.3
;BEFORE_LAYER_CHANGE
G92 E0.0
;7.8

G1 E-2.1 F2700
;WIPE_START
G1 F7200
G1 X74.346 Y19.265 E-.06894
G1 X74.276 Y18.093 E-.41835
G1 X74.303 Y17.061 E-.36772
;WIPE_END
G1 E-.045 F2700
G1 Z7.75 F720
G1 Z7.8
;AFTER_LAYER_CHANGE
;7.8
; printing object soap.stl id:0 copy 0
G1 Z8.05
G1 X148.714 Y6.286 F9000
G1 Z7.8 F720
G1 E3 F1500
M204 P800
;TYPE:Perimeter
;WIDTH:0.899999
G1 F2100
G1 X148.714 Y173.714 E17.44993
G1 X31.286 Y173.714 E12.23878
G1 X31.286 Y6.286 E17.44993
M73 P35 R247
G1 X148.594 Y6.286 E12.22627
M204 P1000
G1 E-2.1 F2700
;WIPE_START
G1 F7200
G1 X148.596 Y8.686 E-.855
;WIPE_END
G1 E-.045 F2700
G1 Z8.05 F720
G1 X149.55 Y5.45 F9000
G1 Z7.8 F720
G1 E3 F1500
M204 P800
;TYPE:External perimeter
G1 X149.55 Y174.55 E17.62411
G1 X30.45 Y174.55 E12.41296
G1 X30.45 Y5.45 E17.62411
G1 X149.43 Y5.45 E12.40045
M204 P1000
G1 X149.388 Y6.233 F9000
G1 E-2.1 F2700
;WIPE_START
G1 F7200
G1 X149.432 Y7.85 E-.855
;WIPE_END
G1 E-.045 F2700
G1 Z8.05 F720
G1 X57.685 Y14.513 F9000
G1 Z7.8 F720
G1 E3 F1500
M204 P800
;TYPE:Perimeter
G1 F2100
G1 X57.756 Y14.708 E.02171
G1 X57.973 Y15.736 E.10942
G1 X58.1 Y16.865 E.11846
G1 X58.133 Y17.967 E.11494
[... 1790 lines cut ...]
G1 X107.214 Y140.924 E.11401
G1 X107.575 Y139.759 E.12718
G1 X107.931 Y139.107 E.07734
G1 X108.412 Y138.483 E.08212
G1 X109.123 Y137.914 E.09498
G1 X109.747 Y137.568 E.07432
G1 X110.92 Y137.198 E.12817
G1 X111.926 Y137.025 E.10637
G1 X112.531 Y136.976 E.0633
G1 X113.568 Y136.939 E.10811
G1 X114.596 Y136.98 E.10727
G1 X115.537 Y137.087 E.09867
G1 X116.447 Y137.269 E.09678
G1 X117.202 Y137.512 E.08261
G1 X117.795 Y137.821 E.06968
G1 X118.104 Y138.022 E.0385
M204 P1000
G1 X118.609 Y137.909 F9000
G1 E-2.1 F2700
;WIPE_START
G1 F7200
G1 X118.554 Y138.373 E-.20322
G1 X118.972 Y138.866 E-.23028
G1 X119.18 Y139.165 E-.12974
G1 X119.478 Y139.699 E-.21784`

amthe, 2023-01-14:
Its happening somewhere here in these lines: (Removed everything with G1...
`; generated by PrusaSlicer 2.5.0+MacOS-x64 on 2023-01-14 at 17:31:01 UTC
;
; thumbnail begin 16x16 804
; iVBORw0KGgoAAAANSUhEUgAAABAAAAAQCAYAAAAf8/9hAAACIElEQVR4AbWTX2vTUBjGc1Vv6k0vSq
; 1U65+WpE2bJulMU5Ka2KqbOP9MW2fXbd0cbKAXIog3KkH0QkULIirotlJnqxcDN5Vd5qPkozye9xRk
; Io564YHDgeR9n+f3vCcRhP+1MocPeJFIpC3864rt35dsFw96vek81loGdF0PwuGwM3Lzk4m0/6B2HI
; /H03h+XsR6U0frjAlJkvw9m8l1/Voet+0k3l3N4XOrgPd1GZ0LEtYaOXSmTaiq+icNudbzMe/1pQw6
; F7O8qcvwP15X0GsW8LWtcZJnu2gSiYT3S+Cee9QnZyqi4rdXshjMFPhereewtVDkJ4k+OpviNA+nTG
; iaFoRCoaSwbKfAcuMLQ/5xo8gLVxvDCCS8vaijy056vtFU8IqRvpiU0J0r4dxJE0LNNnHrVAb32eBI
; /ftiEZ9YYZ/tD8x5c15Fj8XpM6L+jMLFKB69e3pZg+A4DizLwqyrchFS31kqskaNk+wsjXHhARcoDE
; mZyZ3KEcxbIgRRFP1qtYpKpYJx18bdmsgjbc6pGLRUfGMRaLBbCzo2mDMNe7l0CLPlNMYUOeCDjMfj
; HgkQCbsqNMrDSG+msvBXDE6wzQTItW0kMXlCCqjnt+ukiSqKEhCNYRg4bShchGhesljkSs2aLA6n/7
; dFyq7rwrZtVK0SbroSVqxjmNBSiMVinjDKoq/NNE2QEMVy9Cyi0ehozbsjybLsl8tl7PVX/gSRpSV7
; S4ct7wAAAABJRU5ErkJggg==
; thumbnail end
;;
; thumbnail begin 220x124 16512
; iVBORw0KGgoAAAANSUhEUgAAANwAAAB8CAYAAAACRt5vAAAwJ0lEQVR4Ae2diXNVVbr2qeouv26r+/
; u6W29XV9/u1kZUEJlCwAQIhEFAmSeZEhIQRJFZ7XYWh1YDiiDaiAgozRAIGBICiKIgONv3flX3H/BP
; 2Xf93r3evdfZOcPeyTk5AfaqWhU4yT7n5GQ9+3nX8z7vu/r0SUfZxg033HDrH/7why3Dhg3zsk2+1y
; cd6UhH9wdgu/322y9OrKrwFoyr8HKBbuDAgT/xs33SkY50dG0oq62ePNz78pF7ZD41ozIn6Fy2U1a8
; 5ZZbDvA1ZcF0pCPHULDAagBMwaazdcVIj+/lY7uUBdORjhgDQAAMAAWwomBzZyG2a2xs9JYvX57u+d
; KRjugoxGq55r6lIzoBasyYMV5DQ4M3Z84cb/To0TKjP8O+sE860nE9DpfVAFBcsLkhpgoq9913n7dx
; 40Zv7NixAjyd1dXVXm1tbacQ81e/+lVtn3Sk43oZKowkYbVc8x+La736+nphNBds0ZmGmOm4LgfsEj
; eEPPtgPECeerDKq5+SH3Cw3b333puyXTqur3Hrf/y/xkLCiILtuUn9vJem3OGdaqyIBbzn51bnBR0s
; OGHChJTt0nHtjz/8+v/c2lj5n1viAOed2Xd5q6v+IvOpCbd5zxrg7X9gUCzQfbisKi/olO2iokoqqK
; TjmhmA7Vjd0J/isJoLNhdwzH/cd4f3ycrCAkvcEBORJQ0x03FNjbisBtg2j/1rBtiigNMZl+3eWlRd
; kO2yCSo4VfqkIx1X04gbQmZjtUKAY745Y0BRBZVJkyalbJeOq3MM/eP/rY0bQmZjtTiA0xCzWIIKc9
; y4camgko6rZ3RVGOkq4JKGmLDdjAn52W7UqFGdBJXUj5mOXje6I4x0F3DFZjtAN2XKlJTt0tH7RhJW
; O14/rGAI6c6NNX/19i4Y5B1YONh724A0DvCKmT5I/Zjp6FUDsO2cNeBisVlN53YjjJxsqPAMcwpYAV
; 4c0BVbUBk/fnwqqKSjvKO7cn+hicuEa1uWDfOaDeCalw71Tph//2vxkB4PMZk1NTVpiJmOnh+lEEai
; E4CeWTnSO718uDDbkaVDvKMGcICvzTy2dVr/WKBj/nPuwKKx3f333+/NmDEjZbt09MxIIow03X9nl8
; DG3Dt/kDhKYCjCSQUc4WXHikqZcQFXLLajzGfBggXexIkTpeYuZbt0lGwkYTX2WV0FGnPL5NsNi1V4
; p1cM9z42AGu2gOMr//9s9T3eBTP3GXEkCei6I6hQbzd16tSMejuUzFRQSUfRR6nk/mxz3ZhbvUNLhn
; itho0IHU/a/RvsBtO1m8eurBvtfbexRgD54pTbE4MuiR+z4b4aARsulFyCSjRZnoaY6ejyKIUw8uxE
; f9EDLvfxZybeJvstRBGYDJZThRLQtZi9HCD7fPVI77OHRnofJkgVdIftdi4Z1SVBJfVjpiP2KJYP0p
; 3rDcB2G0B98bC5dk2VAOklw1BPG6C9aMLIN2f0l+/DcLAb4ILpEE2YAA4gfrpqpPfVo1XCdlzz9/F9
; uwy6YqcP0gLXdCQeSXyQSPdxwQajdBgQfb9pnPfN+tGGvYYJo71wbz/v5al3eG9MN4CbN1DCR8B2zo
; omhJWqVAKyL9dUez+Y54AB98y7W66P60bpifRBNGeXCirpyDpKJfdvMGA7YMAGcADc5w9XeecNS8FY
; e+bf7b1m1EwA86pZ+O8ZwPH4J6tGeJcerfY6DKjJvR1e4oeZAA7R5LwJKQHsBwsGCVBhye6ALqmg0h
; U/JoJK6sdMh4xSCSMbam41ILpbQARwAAzPAYPBZHsMwADMK4bhYMvtJkQEmFcM2L7dUOOdMT8L2LiW
; 54CJeIwJ671vAPvWrAEGtP71AC8KJFo0bJ85QMLXYrEdIeb6GaMKgg7BJQ0x0xGMJKzWFccIYDq5rC
; JjDwbYWNRHDJAIKbFvvW5YjrnN/Py/Fg+WcBIWI9HNNbqPU5Zjj0eICUO+ZcC0ddqdXpOZ+vUNA9wd
; BogHzXNxDa/NnnHXnJ73Y2YTVNL0wXU4Su2DBJyEkLrYURphrzYDNsQQ1EgAh/ABUHQP5zNclRFGqr
; 2zK/2Qkmt5jhP14fUA7n0HcID1HQMofu6Sed9fr6028n9lkMtj8lzvzhlYFkEl2rwoZbvraEy98z8a
; S8VqzMfG9fWO1fs2LF/wCBc8jwEYQsXdDuD4yh4OgCCqfLdxrDhKACzhZ7Nj7TplAYdoooADsIBJAf
; fvxycYpvQBy/ViDbNhaVxrWLEFlWgPlVRQucZHT/ggmTtn3eV9akLC1sCaNTRwihBSEhYeDkLKTMAd
; qx8qeTbcJG16vflZNS9zvewBzeMw3HYBXH8JI2E4XovENuJMuwN4BZwk1M37SCKoJPFjFhJUcqUPUk
; HlGhs95RhBKOmwoGD/xkIXwBiAAED2ZjAPQAJgLsOxsAEi1yOMBNauJb6Xkv+jYDIB3975mQwH4A4u
; Giyg4vqoF5NrePy8uZ7wtVzpAxTMKOhStrtGRikLRHPl29oahws7ueHccesUwZr19foxAkZCQgHcdJ
; +hdlunCQu7fbkP2GbLkACHxy+tGeVdXjtKXuMD81oCOK43c5e5UXy0aHAgrnzckMmwreb/F0zindcH
; eF2xhqUFrunIOUotjLhz7ehbZMHDUCxsANfieCH5N6/D/uybdaMEPKiMCjhluEPO9Sec6xWwl03i+1
; vzHLDgPgWcZThUSCxf/l6xIgCcXs9jOF0ALCz7zpyBiQGnbBfXj7lsalrgel2MUsj9r913p+yDYAcY
; iaT2o6NukaQzC58wDTZhYbctD93+x6xCqbk4JklrQIa62DStf2DtguFaHYY6Xm8VSkdwYY8HMLmG99
; HkhJQfRhjuuFVJuR6Qt0qJz3B5jL0m1QmlTpa/MG+UpAfyAW/RokVpiHk1jlL4INeNvlUcIQDtvx6r
; 9b4yDMGCJXdGQhmXCOLFHmvNYrF3POiriZr05udhra8Nu327YYwAB0bEmsXzSFrA7uFO2z0cAGuxOT
; z2YYCW1xaGM88PIz45oa8wjgJOqw30ek2an7AWsUtrquR3gOEQXZ7rIti6kj6A7aLA4//U182fP9+b
; PXt2WuB6NY1S+CAVbOynyIuxB7uweqQAiwUPQ2yZ3E+YigUMIxFuYcFqtwteUwIwy5ePjjJhXZVhn2
; EimrxsAPvk+L6iNAI4QMH7Q+UkLQBLATZl1i8eqTLPUS1g3vfA3VLoyvWwL9fzvngd7GNnrfCi7wEA
; fmF+f2xjgPrAwkHyvv9W27dboEsiqLhsR9nP3LlzvcmTJ4srRR+nyDVlu148Sin3wz4sdgDXYU3Fp1
; f4litCSMK55w1LATwAx8L7yoASYKksf1yuN+GkvV6tWbuttQuWozwHAAKOS+zTTB6On2u2DYSQ8nlu
; FErpdVI/LAhLYVnCWsp1AByg+mZDjSS+3aR3m1jLRsgNASCyB8SRgrUMpny2m2wXX1CploLWmTNner
; Nmzcpab5ceONJLRynlfhayG5Kpl/GUTVwTwpFHI6SUsM78m+/h8oCFpC2CvV6TzrCWXq8qJQz68lQ/
; NAUccr0J/U5Z4aRlWaY1DODwc4guAOZ1A3p5DwY4PHbFgP0KSiYMa19bQSt5vOX+Hm7vAv/6Jns970
; FDTG4gsB8iDPvSQ2ZvyfssFtudfmiU9/Dcezud2hoFXdqCvReNUvogN5mfb3esWcesrK6JY+T3d+fe
; JYtSVUYAdMjK+p+uGiGLW8M5f8H7SiF7rEOa+J45IPBRSuJ6iX897AS4YdYMa9jy0IsJI6pKyTSKrL
; wvXuesNTm3WuGl2RFOmM0O4MQaZkJaGPID89gJq6jC6K2O8MJnAbiLyXYcOFJIUMnmUEkLXHtwwGpP
; Teh7oBSsplPFj+PWWiVKo1OJzcIGIJq4ZuGzWLnm+41jJKwEcMcda9ZJR2nM5qUEsLDYd1i7TEiooe
; PRpWHVtwL2qDUvA9htNg+nXkq8mD9uHieho+byXGuXAu59y5DcMLY5ossXZo/634+Pl5vGyYaKjNQE
[... 1266 lines cut ...]
;WIDTH:0.9
;WIPE_START
;WIPE_END;TYPE:Solid infill
;WIDTH:1.06712;WIPE_START
;WIPE_END;TYPE:Internal infill
;WIDTH:0.9;WIPE_START;WIPE_END
;WIPE_START
;WIPE_END
;WIPE_START
;WIPE_END
;WIPE_START
;WIPE_END;WIPE_START;WIPE_END
;WIPE_START;WIPE_END;TYPE:Solid infill
;WIDTH:1.06712;WIPE_START
;WIPE_END;TYPE:Internal infill
;WIDTH:0.9;WIPE_START
;WIPE_END
;WIPE_START;WIPE_END;WIPE_START;WIPE_END;WIPE_START;WIPE_END;TYPE:Solid infill
;WIDTH:1.06712;WIPE_START
;WIPE_END
;WIPE_START
;WIPE_END;TYPE:Internal infill
;WIDTH:0.9;WIPE_START
;WIPE_ENDM73 P40 R229; stop printing object soap.stl id:0 copy 0
;LAYER_CHANGE`

Prusa-Support, 2023-02-04:
Thanks for reporting.

An allocation cell on the SD card could be damaged, rendering the SD card corrupted.
You can try to fix via _slow FAT32 formatting_ or specific tools. The Windows' _Repair Tool_ could work.
The best practice though would be to try other SD cards and other `.gcode` files.
https://help.prusa3d.com/article/sd-cards-and-usb-drives_112291

On a side note, the G-code you provided could be incomplete. Try to "zip" it or simply rename the extension `.txt` then drag and drop the comment box.

Michele Moramarco
Prusa Research

closed 2023-02-22 by jonnieZG