prusa3d/Prusa-Firmware #1405: Regular extruder temperature drops with firmware 3.5.0

chris680, 2018-12-13:
Maybe related to issues  #1362 and  #1368

With update 3.5.0 on my stock Mk3, I got little temperature drops (~10°C maybe more) on the extruder. I noticed those while looking at the octoprint temperature graph (every 30 min or so). I a generally not next to the printer so I can't tell if the temperature drops trigger "hotend temperature runaway" alarms, but the prints succeded. However this seems due to the 3.5.0 firmware update, as I could not replicate this moving manually the printer head (no bad contact) and these drops disappeared as soon as I reverted back the firmware to 3.4.1!

As I could find other users with the same problem on the forum, I thing there is an issue with this firmware, especially as the temperature drop are really close to the delta-T that triggering alarms (15°C if I am not mistaken).

tobymurray, 2018-12-15:
I see this also on 3.5.0 on my stock MK3. Trying to compensate seems to have put me square in #1362 territory:

[image]

You can see the temperature drop when the fan kicks on. Not sure why it drops to 0, but you can see it struggle to maintain temperature. I upped the target temperature to try and compensate and nothing happened, so dropped it back down. That's when I got the PREHEAT ERROR and it aborted the print.

bionno, 2018-12-21:
So with 3.3.1 you did not notice any temperature drop ?

tobymurray, 2018-12-21:
I rolled back to 3.4.1 everything appears to work as expected in that version. I did use 3.3.1 when it was the latest and had no issues. I didn't try 3.5.1, anticipating it still being broken as both these issues are still open.

tobymurray, 2018-12-21:
I rolled back to 3.4.1 everything appears to work as expected in that version. I did use 3.3.1 when it was the latest and had no issues (other than some "time remaining" quirks). I didn't try 3.5.1, anticipating it still being broken as both these issues are still open.

cnlson, 2018-12-25:
@PavelSindler There is a full thread on prusa forums on this can you please investigate?

photog0411, 2019-01-24:
When can this be fixed?

crashtest22, 2019-01-24:
This is also happening to me on 3.5.1 with new machine and flash/reflash.
I even spent several days on the chat with 2 different people before I started looking elsewhere. 
I have even had one CSR send me another thermistor with no change. 
I am VERY surprised the CSR's don't know of this issue that it seems many have had. I added a post on that thread with my temp dips. 
Can we please get this sorted out so I can print on this amazing machine as it was intended to print with the latest firmware :)

avirtuos, 2019-01-30:
Same here. Brand new prebuilt mk3. My first 3 prints we're fine, then the new firmware message popped up. Post upgrade I see 5-10c drops in temp periodically for short periods (usually 10 - 30 seconds)

lord-carlos, 2019-01-30:
I rolled back to 3.4.1. Everything seems to work.

photog0411, 2019-01-30:
So can't they just rollback whatever caused it from 3.4.1 -> 3.5.0

CNCKitchen, 2019-02-03:
Some problem here with my Mk2.5. Just rolled back yesterday to 3.4.1 and everything seems to be fine again:
[image]
[image]

TurboMech, 2019-02-09:
I’ve been printing with my Prusa from 210C to 380C extruder (65w heater and PT100 sensor) and have yet to see this. 

Could this potentially be when your fan kicks up to 100%? This could be an issue with you running in Stealth Mode? Lastly have you completed a PID tune of your extruder?

lord-carlos, 2019-02-09:
> Could this potentially be when your fan kicks up to 100%? 

No, I have a sock and it happens even on PLA prints where the fan is 100% active all the time.

>This could be an issue with you running in Stealth Mode?

I don't run in stealth mode so I doubt it.

>Lastly have you completed a PID tune of your extruder?

Yes, I have done the PID tune.

> Edit: I also don’t use the filament sensor. Could you try disabling that for a print to see if anything changes?

Have not tried it, but the printer just stops with the message "heating" would be weird if that was a fillament sensor issue.

I might even revert to 3.3.1 as I  don't have an MMU anyway and there have been plenty of bugs in the last few releases.

tobymurray, 2019-02-09:
This is nothing to do with anything but the firmware. On my stock MK3, running out of the box PLA profiles not in stealth mode with no filament sensor enabled after thousands of hours of successful prints, 3.5.0 (and 3.5.1) fail obviously. It's not a subtle "maybe it could be..." phenomenon, it's an in-your-face show stopper that's clearly visible on a temperature graph. There's an arbitrary seeming temperature drop, which often leads to clogs. While I've seen many others running into it, I'm surprised _everyone_ isn't blocked by this.

I would imagine a heater that's half again as powerful could hide this. Have you looked at a temperature graph closely to confirm you're not getting irregular temperature dips? I don't mean to be confrontational, but I have a vested interest in this being fixed (I love printing), and I'd hate for a reader to think this issue isn't easily reproducible or well understood.

@TurboMech - completely unrelated, do you have a resource for how to install the PT100 and have everything work out? I'm interested (and the 65w heater too...).

CNCKitchen, 2019-02-10:
> I’ve been printing with my Prusa from 210C to 380C extruder (65w heater and PT100 sensor) and have yet to see this on 3.5.1.
> 
> Could this potentially be when your fan kicks up to 100%? This could be an issue with you running in Stealth Mode? Lastly have you completed a PID tune of your extruder?
> 
> Edit: I also don’t use the filament sensor. Could you try disabling that for a print to see if anything changes?

Definitely a firmware or defaults settings issue. 3.4.1 works perfectly. Did several PID tunes at different temperatures, positions and fan settings without any change in behavior.
This is a pretty serious issue for me at the moment because I had a couple of high-temp prints with expensive filament fail due to the temperature drops.

lord-carlos, 2019-02-10:
Mhh, quick question. With 3.5.X I had **two** issues.

* Temp drops that I see in OctoPrint
* Extruder stopping midt print with \*heating\* message on LCD. Continuing after a few seconds leaving zits and drops behind.

Are those the **same** issue? I have not looked at it in detail before I reverted.

AnHardt, 2019-02-10:
I still miss a temperature plot from RepetierHost with the heater power plot on, or a simple log of the M105 responses. From that one could see if the power is actively dropped by the firmware, or if the temperature drop comes from extern and PID is just reacting to slow.

rennne, 2019-03-11:
Maybe this issue is related to:
[3.5.2 heat bed temp fluctuations](https://github.com/prusa3d/Prusa-Firmware/issues/1542)

darBis, 2019-03-19:
Personally i would not completely rule out hw issue. I had experienced similar smyptoms. Somewhat periodic drops in temperature and occasionaly runaway protection triggered.

In my case it was caused by a faulty wire(!) from manufacturing. There was some sort of brekage of wire. When print head got in the right position for just the right amount of time, weird stuff started to happen. There was no visible damage but an "autopsy" after replacing the cable revealed the root cause, beakage at 1/3 of the cable length. After replacing wires from themistor to printer the issue was gone. Just my experience and solution to issue with similar symptoms.

avirtuos, 2019-03-19:
FYI, I recently updated to the official Prusa firmware (Beta) which has 7x7. I've noticed two issues that went away when i reverted back to the community firmware discussed in this issue. 

1. Layer shifts -I've had one layer shift in the previous 1200 hours of printing. After updating to the official beta w/7x7 certainly models which printed fine previously now have a very similar layer shift 100% of the time. 
2. The bed leveling changed significantly and the first 1cm (exactly) of the left side of my bed was too high no matter what I did with my nylock bed leveling. The only thing that helped was the bed level correction on that side. with the community firmway i didn't need any bed level compensation.

This data is just annecdotal but I'm running more tests to see what gets fixed by reverting, reslicing, etc...

holgerurbanek, 2019-03-29:
@darBis I'm a bit embarrassed by this comment: Do you really think, all those bug-reporters (here and in the prusa-forum) are having a defective contact, all with the exact same issues? And that this defective contact can be cured just by downgrading the firmware? 

Or, might it be more probable, that there is a race condition somewhere within the firmware, possibly triggered by using octoprint?

closed 2023-09-14 by github-actions[bot]

darBis, 2023-11-21:
> @darBis I'm a bit embarrassed by this comment: Do you really think, all those bug-reporters (here and in the prusa-forum) are having a defective contact, all with the exact same issues? And that [...]

Since the issue I was having was dependent on current printhead position, previously unnoticed HW issue could manifest like in user reports simply by change of sampling rate or averaging of a measured thermistor value.

I agree, it is quite unlikely for all of those users to have a bad thermistor wiring. I was merely suggesting a thing to check to confirm a firmware issue.

Have this issue been resolved by any of the reporters? I fixed mine by replacing broken wires. :-)