prusa3d/Prusa-Firmware #1408: MK2.5 firmware build on Windows causes extruder axis values to be shown as "?"

BManx2000, 2018-12-15:
I tried building the MK2.5 3.5.0 firmware (commit 9087645) on Windows according to the instructions. The result was all extruder values showing "?" on the screen instead of the mm values. I also tried the latest commit, cc132ae, and it did the same thing.

I did a build of commit 71826a3 on Linux, with the build.sh script and secondary language support enabled, which worked and did not have the bug.

joecrow, 2018-12-26:
I have this issue but I built it manually on Linux. Using the Arduino IDE and the instructions for windows.
Its only the Z height that shows a ? symbol.

joecrow, 2018-12-26:
Fix my issue its to do with printing floating point numbers. I edited the wrong platform.txt file before compiling. The correct one in Linux is: 
`/home/user/.arduino15/packages/arduino/hardware/avr/1.6.23/platform.txt`

Also before when I said it was only the Z it was actually also X and Y.

3d-gussner, 2019-02-16:
Checkout https://github.com/3d-gussner/Prusa-Firmware/wiki/How-to:-Build-Prusa-Firmware

3d-gussner, 2019-02-21:
@BManx2000 @joecrow I thinks this can be closed as PR https://github.com/prusa3d/Prusa-Firmware/pull/1543 You can use Pf-build.sh in Linux and Windows environment

closed 2019-02-21 by mkbel