prusa3d/Prusa-Firmware #1409: Prusa 3.1 Compiling Issue for MK2S

3DPrintingNoob, 2018-12-16:
I'm trying to compile Prusa firmware 3.1 and when I go to verify I get the following error in the image below. I've tried compiling in both Arduino 1.8.8 and 1.6.9. I'm a firmware noob/coding noob so I'd appreciate some feedback...thanks

[image]

ChrisVahi, 2018-12-18:
Have you followed all of the steps in the readme to the T? There are important steps that can't be ignored in there. Have you moved your electronics variant out of the "Variants" folder and renamed it Configuration_prusa? Z_SILENT in defined in that sketch. Try searching (ctrl-f) for "define Z_SILENT" in Arduino and make sure it can be found by the IDE. I found it around line 247 in Configuration_prusa.h. I see you're using Arduino 1.8.8. You may need to install an older version of Arduino.

3d-gussner, 2018-12-21:
@3DPrintingNoob Try my How-to that might help: https://github.com/3d-gussner/Prusa-Firmware/wiki/How-to:-Build-Prusa-Firmware

Also have a custom MK2/s firmware https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.1.0-001-HFM including a merged communication fix that Prusa approved but never made the compiled hex file available.

You need to use the MK2 branch as this isn't merged yet in the default MK3 branch.

3d-gussner, 2019-10-08:
@3DPrintingNoob There is no update for long time. Is your issue fixed? If so please consider to close this issue.

closed 2019-10-09 by 3DPrintingNoob