prusa3d/Prusa-Firmware #1410: MMU2: Load to Nozzle not lifting Z to a reasonable height before flooding nozzle

kagenokurei, 2018-12-18:
I have just built my MMU2 unit and upgraded my MK2S to MK2.5, so I'm not sure if this issue applies to MK3 users as well.

During my tests, I happened to do a "Load to Nozzle" command via the LCD + knob, while my nozzle is homed in all axes. My Z axis is actually lifted up to 0.35mm after homing since I don't want the heated nozzle sitting too close to my new steel sheet w/ PEI without moving for quite some time and end up making a pit mark at that spot. Due to this distance between the steel sheet and the nozzle, the Load to Nozzle functionality created a blob of plastic instead of just building pressure in the nozzle and possibly causing the gears to strip filament. I was able to lift my Z up further via OctoPrint and release more of the plastic and pressure building up due to the flooding of the nozzle.

I believe this would be a real issue in the event that the Z axis is really homed, which I believe doesn't go to 0, but rather sits at around 0.15mm off the steel sheet. I think that the Z axis should be lifted up a certain height before actually flooding the nozzle.

Mjankor, 2018-12-18:
I also see this when starting a single material print (3.5/1.03). Prior to the prime-line, the extruder extrudes a big blob as the MMU does its loading routine. Filament is already loaded to the nozzle.

mgg4, 2019-03-17:
I have confirmed this is also an issue with 3.7.0-RC1.  Load to nozzle doesn't move the print head to a safe place before extruding.  With the printer homed, it produces a blob of filament which has the potential to glom on to the nozzle and heater block.

I would expect this of a G-CODE command, which presumes that other codes would have been issued to move the head to a safe place.  But from the menus, I expect the printer to take whatever actions are necessary to put the head in a safe location.

JohnnyDeer, 2021-01-18:
This issue is fixed by mentioned pull request #1768 , I´m closing.

closed 2021-01-18 by JohnnyDeer