prusa3d/Prusa-Firmware #1411: Speed overflow on extruder when steps * speed > 65535

mionut, 2018-12-18:
I have steps/mm at extruder = 1970, and when i give an exrude command with speed a little bigger than 33.2 mm/s, the extrude never takes places. If a give a little bigger speed of 40 mm/s the extrude is really slow.
I looked in the source code but i don't find any variable in 2 bytes(wich can take up to 65535) wich controls the speed.
Why i have steps/mm at 1970 ? I use Skelestruder V1 with a stepper motorwith 0.9 deg/step from E3D.

3d-gussner, 2018-12-25:
As you try to use a 0.9 deg/step motor you probably hit the limits of the 8-bit ATmega 2560 and Marlin.
Checkout https://reprap.org/wiki/Step_rates

leefla, 2019-02-02:
I think I might be hitting the same, or related issue.
I'm using a 1.8deg stepper, but also with Skelestruder (3.5:1 ratio), which greatly improves print quality.
When printing at a higher flow rate, on a reasonable length extrude (especially if using a 0.6mm nozzle) it just "randomly" stops mid-print (but I believe it'll happen at around the same spot with the same gcode.
I suspect that as you say it's either overflowing the extrusion rate, or actually trashing memory in the planner buffer.
It's not purely a speed thing, since I've had this happen half-way through printing a 'solid' box (with a hole in the middle) with 20% infill, so there's nothing special about that layer, but I think it's the number of extrude steps in that single command perhaps, or at least the combination of multiple commands in the planner.
We "might" be able to work around this by limiting the volumetric rate in the slicer, but if it is purely the number of extruder steps in the line then that wouldn't help (but then it seems that slowing down the print speed does help, so I suspect that speed/accel is affecting the issue in the planner code)

leefla, 2019-02-02:
The following G-code snippet seems to reproduce the issue 100%
(just run it without filament loaded, or you'll get spaghetti)
```
; generated by Slic3r Prusa Edition 1.41.2+win64 on 2019-02-02 at 14:09:45

; 

; external perimeters extrusion width = 0.61mm
; perimeters extrusion width = 0.65mm
; infill extrusion width = 0.70mm
; solid infill extrusion width = 0.65mm
; top infill extrusion width = 0.60mm
; first layer extrusion width = 0.65mm

M201 X9000 Y9000 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M115 U3.1.1-RC5 ; tell printer latest fw version
M201 X1000 Y1000 Z200 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 S1250 T1250 ; sets acceleration (S) and retract acceleration (T)
M205 X10 Y10 Z0.4 E2.5 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M83  ; extruder relative mode
; M140 S60 ; set bed temp
; M190 S60 ; wait for bed temp
G28 W ; home all without mesh bed level
; G80 ; mesh bed leveling
; G81; fetch result for plugin
M104 S195 ; set extruder temp
G1 Z30 F1000 ; lift above bed a little while extruder heats (in case of oozing)
M109 S195 ; wait for extruder temp

G92 E0.0
M221 S95
M900 K30; Filament gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion

;BEFORE_LAYER_CHANGE

G1 Z19.200
G1 E2.00000 F1200.00000
M204 S1250
G1 F3600

G1 X115.418 Y140.639
G1 X140.639 Y115.418 E3.64288
G1 X140.639 Y112.523 E0.29570
G1 X112.523 Y140.639 E4.06106
G1 X109.628 Y140.639 E0.29570
G1 X140.639 Y109.628 E4.47925
G1 X140.639 Y106.732 E0.29570
G1 X106.732 Y140.639 E4.89743
G1 X103.837 Y140.639 E0.29570
G1 X118.790 Y125.685 E2.15985
G1 X118.076 Y126.009 E0.08015
G1 X117.401 Y126.258 E0.07341
G1 X116.586 Y126.493 E0.08667
G1 X115.755 Y126.664 E0.08667
G1 X114.803 Y126.777 E0.09785
G1 X100.942 Y140.639 E2.00214

M204 S1000

; Filament-specific end gcode
G4 ; wait
M221 S100
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
G1 X0 Y200; home X axis
M84 ; disable motors
G1 E-2.0

```
It'll try to print a few lines, then stop mid-print and just "Echo: Busy processing" continually on the TTY.
This is 100% repro on Prusa 3.5.1 firmware with the minor tweaks for 3.5:1 gear ratio (of which the only relevant one is really step ratio)

leefla, 2019-02-02:
As you can see from the comment I added above, dropping the print speed fixes the issue, BUT having the software be so "delicate" that extruding/moving too fast can cause it to complete freeze up is a pretty hideous bug !

vertigo235, 2019-02-03:
@mionut try changing your micro stepping to 16 and use 985 for e steps instead, you don'y really need that many e steps per mm

mionut, 2019-02-04:
Hello, i resolved the problem like you said, with 16 micro-steps for 0.9 motor, so now i can use a 66mm/s limit. Also to reproduce my issue you don't have to print something, you only have to issue an extrusion command with approx 33.2 mm/s with the hotdend heated(no filament) and see that the speed is really slow or not moving at all, and it is 100% reproducible.

closed 2019-02-06 by mionut