prusa3d/Prusa-Firmware #1418: Manually compiled 3.5.1 source code results in inability to boot MK3

nebbywan, 2018-12-19:
When uploading manually compiled 3.5.1 firmware source code in the form of a .hex file or directly from Arduino 1.6.9, the result is the printer's LCD displaying random characters as soon as the "Prusa Research" screen is complete with the fan at 100%. This occurs both for modified and unmodified builds of 3.5.1, but does not occur at all with 3.5.0 or when using the precompiled 3.5.1 .hex file from the GitHub releases page. 

Please advise if any additional information is needed.

Panayiotis-git, 2018-12-19:
Check 2.c at the compile instructions:
https://github.com/prusa3d/Prusa-Firmware

nebbywan, 2018-12-19:
Yeah, that did it. Glossed over that point months ago and this is the first time it bit me, for some reason. Gonna go hide under a rock in shame now.

closed 2018-12-19 by nebbywan