prusa3d/Prusa-Firmware #1419: Nozzle hit heat bed during first layer calibration (Mk3 3.5.1)

haarp, 2018-12-20:
Hey,

I just had the nozzle leave a nice mark on my steel sheet 😿 

It crashed during first layer calibration. Never had that happen before. Steps (can't verify, one hole in the PEI is enough for me 😉)

- Intend: Re-calibrate my Mk3 after upgrading to 3.5.1
- Run XYZ calibration successfully
- The steel sheet is now on the heatbed
- Start first layer calibration from the calibration menu
- During warmup, went into the menu and used the Live-Z entry to set a value close that what I'd expect it to be later on (-0.800)
- Still during warmup, long-pressed the button to manually raise Z up to around 185mm, to verify that both sides are even
- Warmup finishes, mesh bed leveling starts
- Z lowers until the nozzle is buried in the steel sheet, and further

I suspect either the manual raising of Z, the setting of Live-Z before the actual calibration process started, or both was the reason it broke.

Normally I'd expect it to lower until the PINDA detects the steel sheet, then stop, so nothing bad could possibly happen. However, it seems PINDA was ignored under these conditions.

Note that the printer works fine now. I did not use an incorrect/too-low Live-Z value.

inquisitiveKangaroo, 2018-12-28:
Hi
I don't understand why you pre-set your Live adjust Z value to -800 um...? This value might be slightly different after recalibration. 
I would recommend next time to pre-set the value to -500 um and fine tune the rest after first layer cal. starts for the case your new true offset is <800 um.

haarp, 2018-12-28:
> Hi
> I don't understand why you pre-set your Live adjust Z value to -800 um...? This value might be slightly different after recalibration.
> I would recommend next time to pre-set the value to -500 um and fine tune the rest after first layer cal. starts for the case your new true offset is <800 um.

I didn't touch the hotend, so why would the value be different? Live Z mainly concerns the Z distance between nozzle tip and PINDA, doesn't it?

Anyway, live Z still was the same in the end, so it can't have been the reason for this bug.

3d-gussner, 2020-07-13:
@haarp Is this still an issue? If it has been fixed in a newer firmware version, please close this issue.
[stale]

JohnnyDeer, 2020-08-12:
I'm closing this issue as stale, thanks for understanding.

closed 2020-08-12 by JohnnyDeer