prusa3d/Prusa-Firmware #1420: Selector stepper not moving with MMU2 v1.0.3

VjyedSC0UFMc8z, 2018-12-21:
I upgraded from MMU2 v1.0.2 to v1.0.3 and my selector stepper does not move when pressing the buttons or via the menu. I made it move with an unknown button press, but it's not consistent. I swapped the selector stepper plug into the idle plug and the selector stepper moves fine.

All steppers moved correctly when downgraded back to v1.0.2

inquisitiveKangaroo, 2018-12-21:
> I swapped the selector stepper plug into the idle plug and the selector stepper moves fine.
> 
Hi,
I didn't really understood where did you plug the selector motor... to the five pin plug? O.o

llama, 2019-01-24:
I have a new MMU2 which I just assembled (and immediately flashed v1.0.3) and having the same problems. My selector motor also moves when plugged in to the idler plug (the one for the motor labelled "extruder"). I'll try flashing 1.0.2 now...

llama, 2019-01-24:
Yay found the problem. My filament detector ball bearing was stuck. I removed the FINDA and poked at it, loosening it up and now my MMU2 works! Surprising symptoms for such a problem. Thanks Prusa Live Support (Pavel and Marina), you guys rock!

closed 2021-08-02 by DRracer