prusa3d/Prusa-Firmware #1422: MMU2 - 1.0.3 does not move to the correct position.

dulfe, 2018-12-21:
I just upgraded the MMU2 firmware to 1.0.3 (from 1.0.2) and when I try to load a filament the "head" moves to the wrong position, it is too far to the left (front view) and that causes the filament to get stuck and it does not load. The head no longer aligns with the filament path.

When I turn on the printer the MMU does the initialization (which is really fast in 1.0.3), then I try to load filament #1 and after the "second initialization" it moves the head to the left and it ends flush with the frame.

[image]

[Full initialization and filament load video can be seen here.](https://youtu.be/97GmG4geUBU)

I downgraded to 1.0.2 and the MMU heat moves to the correct position and the filament loads without a problem (not flush, but with a small gap between the frame and the head).

[image]

[Full initialization and filament load video can be seen here.](https://youtu.be/zB0UAYcXxxY)

Is there something I need to do to "align" the head properly?

Daniel.

satang, 2018-12-24:
Same issue here

Panayiotis-git, 2018-12-24:
I think that the problem starts from the homing procedure of the selector. Instead of moving at once to the far right and home for 3 times, sometimes it moves jerky to the right and it does not even touch the end wall.
So, when it moves to the left, it does not align correctly with the first position.

Grismo, 2019-01-06:
Same issue here. Initialization is much faster now since selector is not moving to the right anymore (where the stepper is located). But therefore the noise is much louder when calibrating the selector and, as dulfe described very precisely, that gap is not visible anymore. In my opinion the new calibration position for the selector is not as good as before since the left end of the MMU is not as stiff as the right end (its bending a bit) which might cause the misalignment. I personally would prefer to set the calibration position back to the side where the stepper is located. That seems to be more reliable.

Grismo, 2019-01-06:
After checking the STL-files for the MMU I think that there should be a 1mm gap between the left side of the pulley body and the right side of the selector in order to have the openings for the filament aligned.  But with the new initialization procedure the selector doesn't move anymore after initialization is finished, so the openings are misaligned by 1 mm. May be that the necessary offset has been forgotten to consider in the firmware when changing the initialization procedure.

gooseBot, 2019-01-10:
I had the same issue. I reverted to 1.0.2.

Grismo, 2019-01-10:
For the time being I use a spacer like this one: https://www.thingiverse.com/thing:3343838

dulfe, 2019-01-10:
@Grismo I will definitely test that spacer tomorrow. Thank you!

dulfe, 2019-01-11:
@Grismo I tested it and it worked great!. Thank you for workaround.

TheFitzZZ, 2019-01-30:
No word on the cause of this? Or why not many seem to be impacted? I too had to revert and am wondering why no fix has been released yet. It's been a while.

Panayiotis-git, 2019-01-30:
This is the wrong place for this issue. It should be at the https://github.com/prusa3d/MM-control-01

JohnnyDeer, 2021-01-19:
This isssue is related to https://github.com/prusa3d/MM-control-01/issues/94 and fixed. I´m closing.

closed 2021-01-19 by JohnnyDeer