prusa3d/Prusa-Firmware #1425: Feature request: Multi steel sheet option

danielegger1, 2018-12-22:
I have five different steel sheet (prusa pei, prusa txt, buildtak pei, normal buildtak, thekking txt) and all of those needs different live z values. I have a mmu unit and this provided me the idea of the sheet selection option.

My idea is to have a possibility to add several live z values in the live z settings menu from sheet 1 to sheet 5.
So for a user with just one sheet only sheet 1 will have a value, but cor those useds which has several sheet we can change the value for the second, third etc sheet to the necessary value and if there are more sheets with a value unequal to 0 it will pop up with a message before the print starts with the current live z value - this can be similar to the filament selection message before a print start. Currently I forgot quite often to change the live z before the print start and I find it iut when the first layer is already started, which leeds to aborting the print and start all over again.

To have the offset in slic3r does not make much sense to me as I prepare the next days already much earlier and I swap the sheets depending on the surcafe which I wanna get.

I think this can be a quite helpful feature as also the two original sheet needs a offset of about 200-300 mikrons.

workinghard, 2018-12-26:
Actually Live-Z also depends on the material you're printing, not only the sheet. 
Even better solution will be to have RFID chips in the sheet and the filament roll. Then you can store all the preferred values for this combination. But this is more a wish for MK4 :-)

danielegger1, 2018-12-26:
Haha
Yes and no

If the temp calibration for the pinda is correct you only need different
values for different sheets (and only fine tuning for filaments but this
can be also done with first layer height in slic3r) and to wait for that
simple addition for the MK4 is not a suitable solution I think as also for
Prusa guys it is probably annoying if you switch between sheets. So maybe
you can add that in a future release.

But thanks for the response by the way.

workinghard, 2018-12-26:
That's true, you can fine-tune in Slic3r. But then you could also do it already now for different sheets in Slic3r. 
I don't really see a difference in the process. Selecting manually on the printer the live-z value X from the menu or setting it manually via Live-Z adjust is kind of same to me.

danielegger1, 2018-12-26:
I am not sure if you get what I mean.

The idea is to have for example 5 different values in the settings for 5
different sheets.
And in most cases it does not matter which sheet but in some cases it
matter.
So just before the print start I will get the question “which sheet setting
you want” and maybe also add a last option “remeasure bed” (with this
option I could swap the sheet if I forgot to and will come back to the
question). For some prints the powder coated is not the best option but
maybe on the previous print this was the correct surface what I want to
have.
I am swapping several times a day the sheets and so I forgot to change the
live z also several times and I end up to restarting the print because the
life z is wrong and I will release that when it starts the first layer (as
some are just significantly different)
I could do that in Sclic3r yes, but also the old and the new prusa pei has
different thickness and if I just swap between those two I end up with
changing live z very often and if I have to do that often it is quite
annoying. So said I don’t know when slicing which sheet will be taken.

The big advantage of the message is that I have to double check which sheet
is on and select it, so I cannot forget it.

Another idea (which will be already helpful enough) is to make a Mxxx code
which makes a live z popup window (so I can insert that before the start)
and the printer is haltet until I press the knob, so I can modify the value
before each print, similar to first layer calibration just on a normal
print and the printer does not move in that case.

inquisitiveKangaroo, 2018-12-28:
Hi
Thank you for an interesting feature request. Can you create a pull request? Otherwise I'm afraid this will not be the highest priority for several weeks due to the limited numbers of developers.
Wish you happy new year!
Kangaroo

workinghard, 2018-12-28:
@cnlson: In Slic3r it's under the "Printer Settings" tab -> Custom G-code -> Start G-code.
Place this two commands in the first line:
```
M117 "Make sure sheet tkv3 is in place"
M0
```
Under "General" update the Z offset for this sheet.
Save this profile as your tkv3 profile. Do the same for the Prusa PC sheet.

It's an untested idea. Please let us know if it works this way with the current firmware.

cnlson, 2018-12-28:
@workinghard i'm not on the current firmware. I'm on 3.3.1 which is the last fw that i am comfortable running with all the issues. The strange feed issue with loading filaments and the temp drops in 3.5 and 3.5.1 until the issues are worked out, i'll stay on the sidelines with a working printer. 

so if this is not in 3.3.1 it is academic for me

workinghard, 2018-12-28:
@cnlson Then try it on 3.3.1 and let us know
BTW: Maybe it would make sense to contribute to the issues holding you up to switch to the most recent firmware. Otherwise it will not help you if this new feature will make it's way in the most recent release.

cnlson, 2018-12-29:
I checked after reverting from 3.5 and the issues were logged:

https://github.com/prusa3d/Prusa-Firmware/issues/1392
https://github.com/prusa3d/Prusa-Firmware/issues/1389

i added my comments on both.

on general, i assume you are talking about z-offset? i might play with that a little but my concern is that the difference isn't as large as it seems to expect -0.3 it's -0.035 and i do not want to screw up either my prusa or my other textured sheet and i am not exactly sure which way this works. in other words is - - or is - + in this scenario. so I'll play with it tomorrow. I haven't really used my original pei sheet plate since I got my King v2 sheet so i'll figure live z on that first and work backwards and make a profile for each sheet.

danielegger1, 2018-12-31:
I found another issue with a similar request.
#1336 

But I got another idea in mind, I would like to have a G-Code command to set the live Z height, depending on which sheet is on the printer, so I could build a "logic" in Octoprint with a couple of sensors so Octoprint will send the correct sheet to the printer without manual intervention (MK4 style :D)

stahlfabrik, 2018-12-31:
+1 for live z adjust via gcode!

michalxfanta, 2019-10-11:
In firmware 3.8.0 we implemented Sheet profiles feature.

closed 2019-10-11 by michalxfanta