prusa3d/Prusa-Firmware #1426: Modified fan speed does not stick during printing with PETG

rennne, 2018-12-23:
During a pirnt with PETG I go into the TUNE menu and change the fan speed to 255.
This works.

But after a few minutes the FAN speed is back to slow again.

inquisitiveKangaroo, 2018-12-23:
Hi,
there are separated settings for the first layer and all the others... you can modify it in slicer > filament settings > print fan speed

cnlson, 2018-12-25:
> During a pirnt with PETG I go into the TUNE menu and change the fan speed to 255.
> This works.
> 
> But after a few minutes the FAN speed is back to slow again.

anytime the gcode includes a change your manual change will be overridden. If you want to make a global change you need to make the change in the slicer settings.

rennne, 2018-12-25:
To make the Tune-Menu useful in practical terms, the change should stick till the end of the print.

If I would have set everything perfect in the slicer beforehand, I would not have to “tune” anything.

cnlson, 2018-12-25:
You keep saying tune but I do not think you understand what that means. It's not a global override. It's a temporary change until the next command. If you want a global change you need to put it in the gcode. The gcode is a series of commands. Once one of them changes your setting you'll need to override it again. And can in petg is super variable. If you changed the temp it's likely that will not change until the end of the print (after the second layer) and fan on pla probably doesn't change either but can on petg is variable in the extreme. Speeding up and slowing down. Mainly because it affects bonding and part strength. The more fan the less strength.

rennne, 2018-12-28:
I play the guitar, so I know what tune means: One changes something and that stays that way at least for a few hours.
I assume many people have a similar understanding of that term.

As I said the actual behavior is not very useful, and the naming is misleading. Because why would one use this feature when it is unknown how long is may stick? and the printer is not even showing the actual fan-speed on the info screen

I think people would be better of if it would not exist at all with this behavior, or make at least a global override, or even better to have a fanspeed-factor (for eg. 1.5 would tweak every related gcode with a factor 1.5), or a offset which is always added or subtracted

stahlfabrik, 2018-12-28:
So when playing the guitar and you play one note repeatedly, then this tone keeps playing. But when you follow a melody, your hand grips different notes on different strings. And here the analogy ends.

But your gcode is a sequence of "tones" aka fan speed settings. So if you override the fan speed at one point in time it only stays until the "score" aka your gcode tells the printer to dial in another fan speed.

The tune menu works just fine - it is your gcode who tells the printer to fall back to your "unwanted" fan speed:-)

Imagine you have a print with no fan on and discover that in order to save the print you need some cooling. Now you can enter the tune menu and dial in an absolute number to activate the fan.

I never used the tune menu for altering fan speed - just to make that clear:-)

rennne, 2018-12-28:
I was referring to "tuning" the guitar, and not to playing it. When tuning one changes the sting tension, and if I tune it tone higher, the every note I play afterwards is a tone higher.

cnlson, 2018-12-28:
seriously... this is not an issue. if you wish to override the settings in the gcode then you should do that in the gcode. The machine will interpret instructions as the gcode presents them and if you wish to modify those commands after they are give you will need to do so repeatedly and without complaint. 

you should realize by now that you are making yourself look uneducated in 3d printing. 
to use your analogy, it is as if you are "tuning" a player piano and then are surprised when it plays other notes than the tuning chord. as with a player piano or a record player or cd player. it will play the next note it reads from the media and the printer will follow the instruction it has in the gcode for fan. 

this is not hard to understand so I am beginning to wonder if you are trolling.

maximlevitsky, 2018-12-28:
Actually he has a point. Think about print speed and even flow rate. Both are highly variable through the print, but printer does allow to modify them based on percentage.

This is what @rennne wants I think, and its valid feature request rather that trolling.

I personally only print PLA for now, but eventually will print PETG, and I probably would want such feature as well.

stahlfabrik, 2018-12-28:
Maybe the tune menu for the fan should hardly set the minimum fan speed. So gcode could go faster but not slower. Except in the end where the m107? MUST disable the fan. So the code for that command has to be altered too

workinghard, 2018-12-28:
IMHO the tune setting should never override the gcode instruction. gcode is telling you what is required at time X.  
Maybe it make sense to go the same approach like the global moving speed adjustment. Let the user change the percentage of the current fan speed setting.

cnlson, 2018-12-28:
@stahlfabrik @workinghard The tune menu works fine as it is and as it is expected. If you set a temp prior to the first layer, when it switches to first layer time it will override your setting, if you change a fan setting but the gcode has a following change to the fan it will override. 

Tune doesn't abrogate what the gcode has, it allows you to "meddle in the moment" 

Some changes appear to be sticky only because they do not have changes in the gcode after they are set. examples would be temp after first layer, live z, speed, flow rate, some can change repeatedly, fan (on pet, abs and PC) movement speed those the change will seem not sticky because there are following instructions. 

@rennne 
I'm not trying to insult you and if you took it that way you are reading me wrongly. I would gladly help you find the settings to modify in your slic3r to achieve what you want. But i will confess am am confused on why you would want or need to run the fan at max on pet. Pet at high fan speeds will not bond, so any parts created with high fan will delaminate and/or have very low strength. with clear pet you will end up with a very filmy appearance also. Fan causes the filament to cool too completely to bond. Especially on an mk3 as the fan is powerful enough to keep the nozzle from going into runaway.

rennne, 2018-12-28:
My fan had some issues so I replaced it with a different fan with lower output, since I had to wait for the replacement part from prusa. My prints where ok.
When I changed then to another filament I saw during the print that the overhangs did not look good, so I tuned the fan speed to max to rescue the print. 

Unfortunately the print was a mess because the fan speed change did not stick. Or maybe I should have changed something else, I do not know,  because the fan speed change did not stick.

The point is, that I am quite unexperienced with 3d printing (I have no clue what a "flow rate" is), as probably many others too, every filament behaves a bit different and it would be great if I could tweak the settings on the printer during one print to get it right, instead of having to print a lot of test prints to dial in the correct settings in the slicer.

The Tune menu works as expected for the temperature, but not for the fan speed. And yes I know it meanwhile because the temperature is not changed in gcode, but fan speed is. But from my unexperienced point of view this is just wrong, and therefore a bug

I know where to set the fan speed in the slicer, but that does not help during the print.

I do not want to annoy anyone, I just wanted to share the user-experience of a newbie.

workinghard, 2018-12-28:
@rennne We all try to improve the printer and the printer experience. The tune menu for nozzle temperature and the fan speed works the same way. For example if you change the nozzle temperature during the first layer, you will be surprised on the second one because it will be set to new one. Same thing, if you print multi material. IMHO this functions works exactly how supposed to. My idea is to provide the operator with additional functionality which might fit you better.

@cnlson No need to explain the existing functions again. Me and stahlfabrik even said the same thing like you did. Try to understand my proposal for the new function.

cnlson, 2018-12-29:
@rennne not annoyed at all. confused at first and then perhaps leaking a little frustration when I thought I wasn't explaining it well enough. 

we have all been where you are, believe me, This is my 3rd printer. The first I maybe got 3 prints to complete in 6 months and then while printing it vibrated itself off the workbench i had it on, and it never worked again. Second printer was a knockoff i3 design that had pretty constant problems, the surface was too delicate, the printer would be out of level if you looked at it wrong, the bearings failed, the reseller went out of business (no more spare parts) So believe me, I completely understand. I have also had the print fan fail on my mk3. I had to use a wire toner to find the wire break so i could cut out the break and get the fan back to working while i was waiting for a fan to ship. And the only way I knew what expected was, is that i had the same experience as you. I just had the print fail and went back and edited the settings in the slicer and it worked.

krisrok, 2019-03-20:
what about making it work as a simple multiplier? i prototyped this in another printer's firmware but it was nowhere complete.

so everytime a fan changing gcode comes up we multiply its value by the user speed before actually setting the fan's speed. 

the user speed has to be in the range of maybe 0-300% so we can decide to put the fan on all-time full speed even if the gcode says it should be as low as 33%.

in my opinion it's the best of both worlds.

krisrok, 2023-09-07:
let's keep it open another 5 years :)