prusa3d/Prusa-Firmware #1433: [MK2s] Losing Z-live calibration after every print

Boby71, 2018-12-31:
Just spent three days in finding out about the follwing behaviour:
**If you reset a the calibration data and you're only running only "Calibrate X-Y-Z" and "Calibrate Z", then the live-Z value will change with every print! If you do a "First layer calibration" as well, then the live-Z stays stable.**
This is very annoying and from my point of view a bug - you find some reports about it in the web (Facebook, Pursa forum) already. I know you're not actively developing the MK2s firmware anymore, but this affects many MK2s out there - I ran into this troubles after I did a recalibration after my original Pinda stopped working. Every user running into this issue will be left behind with a useless printer!

I see two possible ways to avoid this annoyance:
1) **Either** you demand a First layer calibration from the user (message in display, printer won't start working) 
2) **Or** you fix the problem that the firmware doesn't store the live-z when the first layer calibration hasn't been done.

I'm using my own live-Z calibration pattern for 1 1/2 years and therefore I did not rund the 1st layer calibration from the menu. I went crazy about the constantly changing live-Z (and I'm talking about huge changes, like -0,550 for the first successful print and -1,200 for the second print just to make it stick)

Thank you!

3d-gussner, 2019-01-08:
Hi @Boby71, as soon you do XYZ cal the the firmware expects you to rerun the 1st layer cal. Z cal doesn't reset the calibration status. Use G87 at the end of your custom 1st layer cal gcode.

I get every time i power up my MK2 a message that i have to run 1st layer cal after, or connect via octoprint.

More info https://github.com/prusa3d/Prusa-Firmware/issues/301 

Hope that helps

closed 2019-09-30 by Boby71

ToshioMagic, 2020-06-23:
I'm having this issue on a mk2.5 with 3.9.0.

chuyskywalker, 2022-04-29:
This thread saved my sanity. Adding 
```
G87  ; will cause the live z adjustments to actually SAVE after the print
G88

```
into my custom calibration gcode file FINALLY made it save the livez adjust values. I thought I was going mad!