prusa3d/Prusa-Firmware #1434: Feature Request: Action commands

danielegger1, 2018-12-31:
I would like to get "Action" commands when some print issue occur, for example crash detection.
In Octoprint it is possible to react on printer "actions" but for that the printer must sent threw the terminal a message like "action:crashdetected". This can be added for several different things like filament runout, power panic, etc. This way I can send a message from octoprint to my phone (via telegram for example) with a picture, or I can pause the print make a alert or whatever I want.

[image]

Send: N21272 G1 X140.372 Y123.808 E0.00075*103
Recv: echo:enqueing "G1 E-1.000 F2700.000"
Recv: echo:enqueing "G1 Z  10.199 F 800.000"
Recv: echo:enqueing "CRASH_DETECTEDX"

NEW = action:crashdetected

Recv: tmc2130_home_enter(axes_mask=0x01)
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:   0 step=30 mscnt= 482
Recv: tmc2130_goto_step 0 29 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Recv: tmc2130_home_enter(axes_mask=0x02)
Recv: echo:busy: processing
Recv:   0 step=18 mscnt= 291
Recv: tmc2130_goto_step 1 17 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
Recv: echo:enqueing "CRASH_RECOVER"
Recv: T:215.2 E:0 W:1
Recv: T:215.3 E:0 W:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Resend: 21242
Recv: ok
Send: N21242 G1 X143.783 Y126.239 E0.03352*101
Recv: ok

danielegger1, 2018-12-31:
I would also like to get a GCode command (M118 for example) which sends just a action:userAction command.
This way I could easily add this Gcode line on q certain layer height to let octoprint send me a message

DanaViolet, 2022-02-22:
I would also like to see this feature implemented. I would like for the printer to pause when the filament runout sensor is tripped, but this does not happen when printing with octoprint.

napter, 2022-03-22:
It would also be great if we could have a menu item to send action commands. Specifically a menu option for HOST_START_MENU_ITEM which would allow you to reprint the last print job directly from the printer without having to go back to octoprint.

closed 2023-08-29 by 3d-gussner