prusa3d/Prusa-Firmware #1436: head behaviour after error

kwis2, 2018-12-31:
I got `bed preheat error` and after it printer stops in random place and head destroy print (after colling printed element  could destroy nozzle?). Shouldn't head be raised a little?
MK3 printer, fw 3.5.1, Octoprint v 1.3.10

roberthunt, 2019-01-03:
It looks like this is a bug in the 3.5.1 firmware, downgrade to 3.4.1 for the moment until it has been fixed. It seems to be related to the jump in bed temperate after the first layer and the fan kicking in.

See #1362

inquisitiveKangaroo, 2019-02-01:
Hi,
the solution has been already merged.
#1460
You can compile it by your own or the new release should be out in a week.

Have a nice printing day.)
Kangaroo

closed 2019-02-01 by inquisitiveKangaroo