prusa3d/Prusa-Firmware #1440: Bug with power panic and filament runout

izzygold, 2019-01-03:
What happens:
After filament runs out and a power failure occurs it is impossible to continue printing

How to reproduce:
Print with not enough filament for the print, the print will pause and ask to enter a new filament, then shut down the power to the printer

What should happen:
The printer should turn on and ask to add new filament and resume printing.

inquisitiveKangaroo, 2019-01-09:
Hi,
thank you for your feedback. We will take a look on it. Would be possible to provide us what is your FW version?
Have a nice printing day.)
Kangaroo

izzygold, 2019-01-09:
FW version 3.5

3d-gussner, 2020-07-13:
@izzygold Is this still an issue? If it has been fixed in a newer firmware version, please close this issue.
[stale]

JohnnyDeer, 2020-08-12:
I'm closing this issue as stale, thanks for understanding.

closed 2020-08-12 by JohnnyDeer