prusa3d/Prusa-Firmware #1443: M117 fails

GWdd, 2019-01-05:
Entering the command
`M117 Hello World `
does nothing to LCD display.

Trying to use the command in startup gcode and the FW parses lines above and below the M117 and ignores the display command.  

```
M117 PINDA Soak Delay
G4 P15000 ; wait another 15 seconds to warm PINDA
```
In this case the display continues rolling the file name being printed from the SD card.

inquisitiveKangaroo, 2019-01-11:
Hi,
Thank you for your feedback. I tried it by myself and _M117_ seems to work totally fine but there is a different problem. The _G4_ command overwrites your '_PINDA Soak Delay_' text. I wanted to exchange the _G4_ for _M860 S45_ (wait till pinda probe reach temperature 45°C), however, it also overwrite the _M117_. I understand the influence on the functionality of M117 command and we'll try to find a way how to evade it.)
Have a nice printing day.
Kangaroo

GWdd, 2019-01-11:
Mr Inquisitor: thanks for the reply.  I suspected some interaction or msg priority to be the cause.  But others I've chatted with have experienced the issue and tried different things and all basically say the command is useless as implemented.  

In my example, since the G4 timer doesn't display anything, why would it overwrite the message I want displayed?  And oddly, the M860 Wait for PINDA command also has not display message attached.  Both commands introduce an unexplained "dead period" that looks more like a crash than an intentional and needed delay.  

ps: I'm old school: the UI should have absolute priority and be instant.  I really hate the lag as I spin the control knob to navigate the UI.  I've selected the wrong thing many many times.  And I've coded UI's and know it doesn't have to be the way it is.

vojtech-pavlik, 2019-06-06:
Pull request #1900 should also be fixing this issue. Confirmation would be appreciated.

PR #3475 (2022-07-04): Make M117 message persistent on LCD when used with M601 or G4
If M117 is called before M601 or G4 then we would like the custom message to
be visible on the LCD screen until the print is resumed.

Change in memory: +16 bytes of flash

Fixes #3316 (M601)
Fixes #1443 (G4)
Fixes #2970 (G4)

Video demonstration (with M601):

* Before (current Master branch): https://www.youtube.com/watch?v=8Mr-1lYwjsk
* After (this PR): https://www.youtube.com/watch?v=A7_uLBHg0qE
commit bf38241a (2022-07-04): Make M117 message persistent when using G4 command

This is also supported in Marlin 2 FW

Fixes #1443
Fixes #2970