prusa3d/Prusa-Firmware #1452: Paused print with "Choose Extruder" Prompt

NicoSchlueter, 2019-01-15:
My i3MK3 paused during overnight printing. The display shows a "Choose Extruder" prompt. I do not have a MMU nor was this a multi-material part. After clicking one of the entries the printer appeared to continue but failed a few layers later. This might have been due to the blob that oozed out of the extruder overnight where it paused. The screen showed the same "Choose Extruder" prompt again. 

Info:
Original Prusa i3MK3
3.5.1-1778
Standard 0.2mm PLA Profile
Sliced with Slic3r PE 1.41.2+

Changed settings:
Speed changed to 70 (to avoid ghosting)
Stealth Mode
Filament sensor ON

[image]

[Corner_0.2mm_PLA_MK3.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/2759235/Corner_0.2mm_PLA_MK3.gcode.zip)

mdwelsh, 2019-01-15:
I wonder if this is the same root cause as https://github.com/prusa3d/Prusa-Firmware/issues/1455
(different message but same behavior).

inquisitiveKangaroo, 2019-01-16:
Hi,
Thank you for feedback. Could you provide us the gcode?

NicoSchlueter, 2019-01-16:
> Hi,
> Thank you for feedback. Could you provide us the gcode?

gcode is in the attachment "Corner_0.2mm_PLA_MK3.gcode.zip" (link below the picture)
I do not remember what layer it happened on.

inquisitiveKangaroo, 2019-01-18:
> > Hi,
> > Thank you for feedback. Could you provide us the gcode?
> 
> gcode is in the attachment "Corner_0.2mm_PLA_MK3.gcode.zip" (link below the picture)
> I do not remember what layer it happened on.

Thx, I guess I am blind. ^^

rmccaskill, 2019-04-24:
I had this happen to me twice. Once last night. Prusa i3MK2S Firmware 3.1.0. Different files each time though using Octoprint print each time. I was also using Octolapse though I don't know if I was the 1st time or not.

The extruder and the heated bed were kept up to temperature for hours until I noticed the pause!

Selecting "Extruder 1" ended up displaying the same "Choose Extruder" message again. I clicked "Extruder 1" a couple more times, then the Prusa re-homed and smashed through the print.

When I tried to use the settings menu to raise the z height up, this message appeared 

> "Connect printer to monitoring or hold the knob to continue".

(I have a photo if needed)

I held the knob down and a progress bar filled up along the bottom and after the bar was full, the message went away. After that a new option was in the menu system about "Farm Mode". I have not idea if this is related to the 'pause' or not. I added it here just in case.

Happy to share my gcode also if you'd like it. I also have an Octoprint time lapse right up to the time the pause happened.

Let me know if there's anything else I can do to help.

rmccaskill, 2019-04-24:
I tried the same print last night. Exact same problem at the same place. I did use Octoprint 1.3.9
 again this time but I did NOT use Octolapse.

I've attached the gcode here.

[4trees.zip](https://github.com/prusa3d/Prusa-Firmware/files/3111988/4trees.zip)

closed 2023-09-14 by github-actions[bot]