prusa3d/Prusa-Firmware #1455: 3.5.1 - Print suddenly stops with "Press knob to preheat nozzle and continue."

mdwelsh, 2019-01-15:
I started a long print job this morning and all was going well for the first few layers. After a while, however, the printer paused unexpectedly with the message "Press knob to preheat nozzle and continue." This is on a print that I have done successfully before on pre-3.5.1 firmware; this behavior seems to be new in 3.5.1. Is this expected? What would cause this and is there a workaround (other than having to manually press the button)?

Shackmeister, 2019-01-16:
I got this as well today

mdwelsh, 2019-01-16:
I dug through the code a bit and see that this only triggers (I believe) when an M600 command is received. There is no M600 in the gcode I am using -- however, in fsensor.cpp:fsensor_update() I see an M600 is enqueued when the filament sensor notices a runout of reaches an error count threshold. Has anything in this logic changed recently? (Looking at the code it seems to not have changed since 3.5.0, which works fine for me.)

Shackmeister, 2019-01-17:
I'm now downgrading, since I had this issue 8-10 times over the last day. Let me know if I can help with testing or something

xoxota99, 2019-01-21:
Started having this issue today. Any update?

Shackmeister, 2019-01-21:
I ran a calibration wizard and now it seems better. (error wise, prints does not seem better)

bsrotten, 2019-01-22:
Mine does it on certain filaments only.

mujido, 2019-02-09:
I had this problem and eventually noticed it was whenever I was printing with a shiny filament like gold silk. I eventually figured out it was false triggering on the filament sensor. If it is the same for you, you can disable the filament sensor when using problematic filaments.

xoxota99, 2019-02-09:
Same solution as @mujido above. If / when it happens, go into the "Tune" menu (during print) and disable the filament sensor. Note that auto-load and filament out detection won't work with the sensor disabled.

Confusingly, this setting is located in a completely different menu (under "settings") when not printing...

bobek, 2019-02-16:
I had similar situation, also caused by filament sensor, but in my case cleaning up the sensor fixed the issue. (A couple blows of compressed air did the trick)

3d-gussner, 2020-09-22:
@mdwelsh Is this still an issue with newest firmware? If not please close this issue.
[stale]

bobek, 2020-09-22:
> @mdwelsh Is this still an issue with newest firmware? If not please close this issue.

I am currently on 3.9.0 and have not seen the issues for a long time. (I am not the author, so cannot close)

mdwelsh, 2020-09-23:
I have not experienced this issue in a long time. Feel free to close.

closed 2020-09-29 by JohnnyDeer

jsharf, 2021-05-04:
Just wanted to report that I experienced this on 3.9.3. It was near the end of a print and persisted for quite a while. I was out for 1-2 days so it must have persisted the whole time... very loudly. Sorry neighbors.

S3D-r, 2021-11-20:
I also got this same problem yesterday, and since then it has kept stopping the prints. I have the filament sensor disabled, so, a dirty sensor shouldn't cause the issue. Any ideas?