prusa3d/Prusa-Firmware #1457: Temperature ruanway

angelshtilianov, 2019-01-16:
Hi there.
Using the printer in cold environment causes bed preheat errors.
Scrubbing through the code I've found the temp_runaway_check function (commit b14507458994:temperature.cpp:1161) which will monitor every 2 seconds each temperature channel.
While we are preheating and the _current_temperature < _target_temperature (line 1212)
it will check the bed for 2 degree increase every 32 seconds and if there are no 2 degrees gained in 2 error retries x32 seconds attempts (it might gained one degree and we want 2 more) there is temp_runaway_stop, which makes 3 degrees in 2x32=64 seconds for the bed.
Meanwhile we are doing the same for the extruder, but every 2 seconds we are incrementing a counter going to 8. So every 16 seconds we expect the extruder heater to gain 2 degrees which doesn't seem to be in issue, or at least it gains temperature much faster.  But we allow 5 sequential errors on the extruder heater, which means at least 3 degrees in 5x16=80 seconds.
Doesn't it makes sence to allow 5x32=160 soconds/3deg for the bed and 2x16=32 seconds/3 degrees for the extruder insed (generally swapping the 2 and 5 on line 1234) like that:
-if (__preheat_errors[_heater_id] > ((_isbed) ? 2 : 5))
+if (__preheat_errors[_heater_id] > ((_isbed) ? 5 : 2))

angelshtilianov, 2019-01-18:
I've got it working.
I found I have floating point printf issue , but the preheat is working like a charm.
Here I am attaching my patches. Please let me know your opinion.
[0001-Relax-the-required-preheat-temperature-gradient.patch.txt](https://github.com/prusa3d/Prusa-Firmware/files/2772732/0001-Relax-the-required-preheat-temperature-gradient.patch.txt)
[0002-Rework-temerature-display-format-add-float-to-2-deci.patch.txt](https://github.com/prusa3d/Prusa-Firmware/files/2772736/0002-Rework-temerature-display-format-add-float-to-2-deci.patch.txt)
Generally the second patch is just a display formatting change, but the added function is useful when printing floats without actually having vprintf FP lib (I have issues with that part). As I could see there are about 40 places, where floats are printed, so if they get replaced the whole FP lib maybe excluded ? That should free up some flash. I'm not sure about the performance difference between what we have here, and the vprintf implementation.
Tonight I'll try to fixup the floating point issues and I'll provide a hex file when done.

stevek1959, 2019-01-31:
Your firmware worked great with my using Slicr Prusa Edition 1.41.2.   Although when I sliced using Simplify 3D 4.1 , it had showed the Preheat Error at the same layer of printing.  Rather strange?

angelshtilianov, 2019-01-31:
That's a bit strange. I was able to print on a negative ambient using it - it is a quite relaxed in it's requirement for temperature gain. Could you upload your gcodes and let me know your printer model. 
BR,
Angel

stevek1959, 2019-02-01:
It works well on S3D also, my error was I didn't power cycle the printer after installing your firmware.  So all is well, thank you.  You saved me a lot of time.

inquisitiveKangaroo, 2019-02-01:
Hi,
the solution has been already merged.
#1460
You can compile it by your own or the new release should be out in a week.

Have a nice printing day.)
Kangaroo

closed 2019-02-01 by inquisitiveKangaroo

angelshtilianov, 2019-02-12:
Hi there - I've just flashed v3.5.2 on my my printer ...
BED PREHEAT ERROR...

angelshtilianov, 2019-02-14:
Hi there,
I've ported my previous changes to MK3 v3.5.2. (No min temp, relaxed temperature gradient requirement, 0.1 degree display on the info screen)
So far (a day of printing) I have no issues with it.
I am posting a zip with the diff on top of the official MK3-v3.5.2 tree and a hex file built if someone needs it:
[MK3-v3.5.2-Angels.zip](https://github.com/prusa3d/Prusa-Firmware/files/2863721/MK3-v3.5.2-Angels.zip)
Any feedback will be appreciated.